cartographer-project/cartographer #1627: Map suddenly tilts over 90 degrees when everything is still

jyp0802, 2019-08-28:
Hey guys!
I've been using cartographer on ROS for just a few months now and a while back I found some weird behavior with my system of cartographer.

Im running Cartographers 3D SLAM using a VLP 16 and a Microstrain 3DM-GX3® -25 IMU.
They are fixed on top of a turtle bot that is stationary.

But when I run cartographer, whens it roughly 45 seconds into the SLAM running, even though everything is still, the whole map shifts and tilts completely. (Pictures below)

This is when its running well.
[image]

And then it suddenly tilts.
[image]

I've uploaded a **video** and the **bagfile** in this [link](https://drive.google.com/drive/folders/1U2tpxRqyBGyKcFqvDHywKvVnqr9uTLFQ?usp=sharing).

This happens with and without any specific configurations in the lua file.
(For the video one I didn't specify any configuration parameters)

If any of you ran into a similar problem or know what wrong, I'd be super thankful if you help!

Thanks so much!

shamengjun, 2019-09-19:
I am doubt about your imu data is drit at that time, firstly please check imu data at that time.

qmx066, 2019-09-19:
recheck calibration configuration in URDF file first

Qiuxiaochang, 2020-01-15:
Hello, I sometimes have this situation during the playback of the bag, how do you solve it?