cartographer-project/cartographer #1629: How to account for updated estimates of landmarks in map

realdealneil, 2019-09-11:
I am not sure if this is entirely a cartographer question, but I'd like to pose it here anyway, because I'm guessing the community will have a good answer.  Here's the problem: 

We map an unknown environment using 2D cartographer.  As we travel the area, we are looking for landmarks in the environment (objects of interest).  When we recognize a landmark, we estimate the landmark's position in the map.  We store this position.  

Later, down the road, we travel and see more the area and cartographer performs loop closures, improving the overall estimate of the robot's position, and of the whole arrangement of the map.  However, our estimate of the landmark location is in coordinates before the loop closure, so now the landmark position is wrong.  How can we allow the landmark position to be updated when cartographer updates the map?

Yu-Xiaoxian, 2019-09-16:
The poses and landmarks will be adjusted concurrently by bundle adjustment in backend.

realdealneil, 2019-09-16:
Thanks for the reply.  The problem is that I am not using cartographer's landmark system...is that what I should do so that the object's position will get updated by cartographer?  Do I create a cartographer landmark at that location?