cartographer-project/cartographer #1631: If the scene expands even bigger， how can I add a new map based on the original map?

huangshuqing, 2019-09-17:
Hello, every one!
If my robot has been working on a scene for a long time，but the scene expands even bigger one day, so I want to add a new map based on the original map? How to do it is the most reasonable and convenient way？
I am looking forward to your answer.

qmx066, 2019-09-17:
hi, to my experience, you can use your old pbstream file as input statefile, and do mapping, the new generated pbstream file will contain all trajectory information in old state file and align new data with old ones. remember to check frozen_state.

shamengjun, 2019-09-19:
@huangshuqing  hi. I am also interested in this problem, recently I am researching about this problem, finally, I think cartographer could solve this problem, but don't know how to do it, it's my pleasure for me to communicate with you to handle this.

huangshuqing, 2019-09-19:
@shamengjun hello，I have solved it. just as qmx066 said.

huangshuqing, 2019-09-19:
@qmx066 Thanks!

LHZ-code, 2023-04-26:
您好，我在使用这个功能的时候出现了一个这样的问题
F0426 15:30:24.104884 351734 proto_stream_deserializer.cc:72] Check failed: pose_graph_.pose_graph().trajectory_size() == all_trajectory_builder_options_.all_trajectory_builder_options() .options_with_sensor_ids_size() (1 vs. 2) 
*** Check failure stack trace: ***
    @     0x7eff974b71c3  google::LogMessage::Fail()
    @     0x7eff974bc25b  google::LogMessage::SendToLog()
    @     0x7eff974b6ebf  google::LogMessage::Flush()
    @     0x7eff974b76ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x5564017d6518  cartographer::io::ProtoStreamDeserializer::ProtoStreamDeserializer()
    @     0x5564017c58d5  cartographer_ros::(anonymous namespace)::Run()
    @     0x5564017c40dd  main
    @     0x7eff96db9083  __libc_start_main
    @     0x5564017c57be  _start
就是我先进行纯定位，然后结束纯定位的轨迹，然后在开启一个新的建图轨迹，最后会导致pbstream中只有两个轨迹，却有三类传感器数据，请问这个怎么解决的