cartographer-project/cartographer #1644: Updating free space

gullermo, 2019-10-31:
Hello,

Firs of of all,thanks for your great job.
I have been working with the cartogrpher for a while, specially with the 2D SLAM that works perfectly fine.Now I am trying to get my system working for outdoors navigation. For the moment everything seems to be working fine, but I encountered a problem that I couldn't solve. 

I am not being ablo to update de free space of the map, I am using a lidar, and for example when I it detects someone it includes it within the map (which is totally correct),however when that person is gone and the space is free again, it is not removed from the map, so when the robot tries to navigate it encounters obstacles that are actually free.

Is there any parameter that could solve the problem? What am I missing?

Any comments are more than welcome!

Thanks a lot for your help! 

Regards

shamengjun, 2019-11-05:
Theoretically, when the person is gone, catographer will update the submaps. if your robot is still, this maybe update some slowly!

gullermo, 2019-11-08:
Yes, it does happens with the 2D SLAM, but not with 3D SLAM, so I don't know what could be causing this issue... Any ideas?

Thanks for your answer @shamengjun

kdaun, 2019-11-08:
Take a look at https://github.com/googlecartographer/cartographer/blob/bcd5486025df4f601c3977c44a5e00e9c80b4975/configuration_files/trajectory_builder_3d.lua#L73
To keep the computation effort reasonable the free space update is by default limited to two cells.

gullermo, 2019-12-03:
Thanks @kdaun , I will take a look at it and report the outcomes!

closed 2020-10-09 by wohe