cartographer-project/cartographer #1651: question about imu tracker's gravity vector

weidezhang, 2019-11-14:
Hi, 
I'm still confused about how gravity vector was updated in IMU tracker's AddIMULinearAccelerationObservation function. 

  gravity_vector_ =
      (1. - alpha) * gravity_vector_ + alpha * imu_linear_acceleration;

As far as I understand, gravity_vector represents the gravitational direction in tracking frame. confused about why imu_linear_acceleration can add with gravity_vector ? what does the result mean ?

ojura, 2019-11-14:
This is a simple first-order exponential smoothing filter which slowly pulls the value of `gravity_vector_` towards the value of `imu_linear_acceleration`. It is usually used when your measurements are noisy (like `imu_linear_acceleration` are), but you want to follow the moving average of those measurements.

 See https://en.wikipedia.org/wiki/Exponential_smoothing

weidezhang, 2019-11-14:
thanks. i guess it also assumes the imu_linear_acceleration is mostly due to gravity instead of motion ?

garlinplus, 2019-12-02:
> ments are noisy (like `imu_linear_ac`
void ImuTracker::Advance(const common::Time time) {
  CHECK_LE(time_, time);
  const double delta_t = common::ToSeconds(time - time_);
  const Eigen::Quaterniond rotation =
      transform::AngleAxisVectorToRotationQuaternion(
          Eigen::Vector3d(imu_angular_velocity_ * delta_t));
  orientation_ = (orientation_ * rotation).normalized();
  **gravity_vector_ = rotation.conjugate() * gravity_vector_;**
  time_ = time;
}

Every imu measure step,the gravity vector is converted its frame,is it right? If there is a filter for imu acc data,it should be filtered in robot imu frame or in earth frame.But the frame in this is always transformed between robot imu body frame and earth frame.

kyo055, 2021-04-22:
> Every imu measure step,the gravity vector is converted its frame,is it right? If there is a filter for imu acc data,it should be filtered in robot imu frame or in earth frame.But the frame in this [...]

I just want to konw why alhpa equals exp(-delta_t/im_gravity_time_constant_) ???