cartographer-project/cartographer #1653: Compilation error in cartographer/mapping/3d/hybrid_grid_test.cc

kunaltyagi, 2019-12-02:
Current master bcd5486025df4f601c3977c44a5e00e9c80b4975 on compilation from source gives compilation errors regarding missing `value_type` in `hybrid_grid.h`. The errors seem correct since all the `Iterator` classes are missing a `value_type` as required by `std::iterator_traits<Iter>`

## System Details
* OS: Linux
* Compiler: GCC 9.2
* Build Environment: catkin build, no environment flags for `cxx_flags` or `c_flags`

## Error

```c++
In file included from /usr/include/c++/9.2.0/map:61,
                 from build_tree/include/ceres/ordered_groups.h:34,
                 from build_tree/include/ceres/ceres.h:59,
                 from build_tree/include/cartographer/common/math.h:25,
                 from build_tree/include/cartographer/mapping/3d/hybrid_grid.h:28,
                 from source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:
/usr/include/c++/9.2.0/bits/stl_map.h: In instantiation of ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’:
source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9.2.0/bits/stl_map.h:273:4: error: no matching function for call to ‘std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique(cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&, cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&)’
  273 |  { _M_t._M_insert_range_unique(__first, __last); }
      |    ^~~~
In file included from /usr/include/c++/9.2.0/map:60,
                 from build_tree/include/ceres/ordered_groups.h:34,
                 from build_tree/include/ceres/ceres.h:59,
                 from build_tree/include/cartographer/common/math.h:25,
                 from build_tree/include/cartographer/mapping/3d/hybrid_grid.h:28,
                 from source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:
/usr/include/c++/9.2.0/bits/stl_tree.h:1109:2: note: candidate: ‘template<class _InputIterator> std::__enable_if_t<std::is_same<_Val, typename std::iterator_traits<_InputIterator>::value_type>::value> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_insert_range_unique(_InputIterator, _InputIterator) [with _InputIterator = _InputIterator; _Key = Eigen::Matrix<int, 3, 1>; _Val = std::pair<const Eigen::Matrix<int, 3, 1>, float>; _KeyOfValue = std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
 1109 |  _M_insert_range_unique(_InputIterator __first, _InputIterator __last)
      |  ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9.2.0/bits/stl_tree.h:1109:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9.2.0/bits/stl_tree.h: In substitution of ‘template<class _InputIterator> std::__enable_if_t<std::is_same<std::pair<const Eigen::Matrix<int, 3, 1>, float>, typename std::iterator_traits<_Iterator>::value_type>::value, void> std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique<_InputIterator>(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator]’:
/usr/include/c++/9.2.0/bits/stl_map.h:273:4:   required from ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9.2.0/bits/stl_tree.h:1109:2: error: no type named ‘value_type’ in ‘struct std::iterator_traits<cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator>’
/usr/include/c++/9.2.0/bits/stl_map.h: In instantiation of ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’:
source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9.2.0/bits/stl_tree.h:1118:2: note: candidate: ‘template<class _InputIterator> std::__enable_if_t<(! std::is_same<_Val, typename std::iterator_traits<_InputIterator>::value_type>::value)> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_insert_range_unique(_InputIterator, _InputIterator) [with _InputIterator = _InputIterator; _Key = Eigen::Matrix<int, 3, 1>; _Val = std::pair<const Eigen::Matrix<int, 3, 1>, float>; _KeyOfValue = std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
 1118 |  _M_insert_range_unique(_InputIterator __first, _InputIterator __last)
      |  ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9.2.0/bits/stl_tree.h:1118:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9.2.0/bits/stl_tree.h: In substitution of ‘template<class _InputIterator> std::__enable_if_t<(! std::is_same<std::pair<const Eigen::Matrix<int, 3, 1>, float>, typename std::iterator_traits<_Iterator>::value_type>::value), void> std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique<_InputIterator>(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator]’:
/usr/include/c++/9.2.0/bits/stl_map.h:273:4:   required from ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
source_tree/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9.2.0/bits/stl_tree.h:1117:16: error: no type named ‘value_type’ in ‘struct std::iterator_traits<cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator>’
 1117 |  __enable_if_t<!__same_value_type<_InputIterator>::value>
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

ojura, 2019-12-02:
I can confirm that building on e.g. Ubuntu 19.10 with GCC 9 fails. As a temporary mitigation to get it to build, you can disable building tests.

kunaltyagi, 2019-12-03:
Only this test is faling. As such for mitigation, I changed the file from `hybrid_grid_test.cc` to `hybrid_grid_test.cpp`.

I thought the issue would be resolved by adding `using value_type = ValueType;` in `DynamicGrid::Iterator` but no. I thought the class is running afoul of the complicated SFINAE rules for `std::iterator_traits` but adding all 5 required aliases also did nothing. So.... :man_shrugging:

jhbruhn, 2020-01-16:
This fails for me as well. Is there a fix in sight, other than disabling the test?

MichaelGrupp, 2021-03-15:
In the meantime there was this patch: https://github.com/cartographer-project/cartographer/pull/1704

I guess this addresses the issue outlined here, because I haven't encountered build errors in Ubuntu 20 / gcc 9.3.0 since then.

closed 2021-03-15 by MichaelGrupp