cartographer-project/cartographer #1662: Error running demo_backpack_2d.launch

seanxu112, 2019-12-20:
I am trying to test cartographer_ros on my simulation environment. Cartographer compiles correctly on my pc. However, when I tried to run the given demo launch file: demo_backpack_2d.launch; this is the log:

`roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag
... logging to /home/nvidia/.ros/log/3223950a-2293-11ea-9e22-a0a4c558adb8/roslaunch-gamerbox-9400-22524.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/nvidia/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://gamerbox-9400:37751/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES

/
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [22534]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3223950a-2293-11ea-9e22-a0a4c558adb8

process[rosout-1]: started with pid [22547]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [22555]
process[cartographer_node-3]: started with pid [22567]
process[cartographer_occupancy_grid_node-4]: started with pid [22573]
process[rviz-5]: started with pid [22585]
process[playbag-6]: started with pid [22594]
[ INFO] [1576782619.276785901]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1576782619.276979938]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1576782619.277004961]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found 

'/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1576782619.277043939]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1576782619.277066861]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1576782619.277150114]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1576782619.277171897]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.

[ INFO] [1576782619.277206728]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1576782619.277227794]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1576782619.277307488]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1576782619.277330550]: I1219 11:10:19.000000 22567 configuration_file_resolver.cc:41] Found '/home/nvidia/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.

[ INFO] [1576782619.282591149]: I1219 11:10:19.000000 22567 submap_2d.cc:187] Added submap 1
[ INFO] [1576782619.282641101]: I1219 11:10:19.000000 22567 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1576782619.717132024, 1432647016.502781965]: I1219 11:10:19.000000 22567 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635682438164904058'.
[cartographer_node-3] process has died [pid 22567, exit code -11, cmd /home/nvidia/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/nvidia/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/nvidia/.ros/log/3223950a-2293-11ea-9e22-a0a4c558adb8/cartographer_node-3.log].
log file: /home/nvidia/.ros/log/3223950a-2293-11ea-9e22-a0a4c558adb8/cartographer_node-3*.log`

My current protobuf version is 3.4.0, I do not know if this is useful or not. Rviz just shows a frozen screen of laser scan, which is attached below. Is there any suggestions on this problem?

Thanks in advance.

TvdSmagt, 2020-03-23:
I am having the same issue, did you find a solution yet?
I have Ubuntu 18.04 installed, pcl 1.8.1 and ros-melodic-pc 1.7.0

MichaelGrupp, 2020-10-17:
Exit code -11 is most likely a segfault. Did you try to enable core dump and checking with GDB where this happens?

closed 2021-03-15 by MichaelGrupp

lichao951787328, 2022-05-04:
I have met the same problem, I am using ubuntu20 and ros-noetic？