cartographer-project/cartographer #1667: Hallway rotates

spector5, 2020-01-07:
I've recently been using cartographer for localization testing and I have seen an issue on multiple maps that I tested. When going down a narrow hallway, usually not in the first pass through the area, everything will look fine until suddenly the whole hallway rotates (as seen in the image below, compared to the map generated from the mapping operation):

[image]

I have been able to tune the localizer so that the map will eventually fix itself, but I am interested in preventing this issue from arising in the first place. From what I can tell it seems to be heavily influenced by the motion filter, but I have not been able to determine the root cause of the issue.

Here is a link where you will find the config and bag files that I have been working with
https://drive.google.com/open?id=1gOdvxIwmYqBTH9R--FGvBOWrE4GRrR9v

I would appreciate any help you can give in figuring out the cause of this issue.

MichaelGrupp, 2020-10-17:
One thing you can try is to deactivate global localization and switch to local constraint search only, with a prior pose estimate via [/start_trajectory](https://github.com/cartographer-project/cartographer_ros/blob/6296d41cd5a2a82bcb9d23e628ce02f0435c9006/cartographer_ros_msgs/srv/StartTrajectory.srv#L17)

charon-cheung, 2021-01-12:
> One thing you can try is to deactivate global localization and switch to local constraint search only, with a prior pose estimate via [/start_trajectory](https://github.com/cartographer-project/cartographer_ros/blob/6296d41cd5a2a82bcb9d23e628ce02f0435c9006/cartographer_ros_msgs/srv/StartTrajectory.srv#L17).
> Global localization tends to introduce some weird behavior in my experience.

You mean set `optimize_every_n_nodes = 0` ?

soorajanilkumar, 2021-01-29:
Hi 
I would like to know how to disable global localization too. 
Is it by giving
global_sampling_ratio = 0 
or
global_constraint_search_after_n_seconds = 0 
or both?

MichaelGrupp, 2021-03-11:
@soorajanilkumar `global_samplin_ratio = 0` and to be on the "safe" side, set `global_constraint_search_after_n_seconds` to a very high value.

closed 2021-03-15 by MichaelGrupp