cartographer-project/cartographer #1674: Issue restarting trajectory using pbstream and mutable_initial_trajectory_pose

glucas350, 2020-02-26:
I am attempting to reinitialize cartographer mapping using a saved .pbstream file and setting a new starting position for the robot.  The trajectory builder does not seem to initialize fully when using the InitialTrajectoryPose option.  Below is the pertinent snippet of code for initialization, I am not using the ROS implementation.

```
TrajectoryBuilderOptions trajectoryOptions;
::cartographer::transform::Rigid3d::AngleAxis rotation(heading,
          ::cartographer::transform::Rigid3d::Vector(0.0, 0.0, 1.0));
        ::cartographer::transform::Rigid3d::Vector initPose
          = ::cartographer::transform::Rigid3d::Vector(
            x * cInches2Meters,
            y * cInches2Meters, 0);
        cartographer::transform::Rigid3d relativePose
          = ::cartographer::transform::Rigid3d(initPose, rotation);

::cartographer::mapping::proto::InitialTrajectoryPose pose;
pose.set_to_trajectory_id(0);
        // Must use normalized quaternion for rotation
        // cartographer::transform::Rigid3d relativePose
        //  = cartographer::transform::Rigid3d(position, rotation.normalized());
        *pose.mutable_relative_pose() = cartographer::transform::ToProto(
          relativePose);
        pose.set_timestamp(0);
        *trajectoryOptions.mutable_initial_trajectory_pose() = pose;

// Add a new trajectory
AddTrajectory(trajectoryOptions);
```

Here are 3 images describing what I am seeing.   The blue circle represents the position of the robot and the red is showing the 2D lidar scan.

This is position  x:0, y:0, heading:0 in the map. 
[image]

This is where I am attempting to set the position to using the above code snippet.
[image]

This is the result.  The robot position seems to be correct, but the range data is not in the correct location.
[image]

I am getting the robot position using the LocalSlamResultCallback()
```
<const LocalTrajectoryData::LocalSlamData>trajectory_data;

// Robot Position
::cartographer::transform::Rigid3d position = pose * trajectory_data->local_pose;
  x = position.translation().x();
  y = position.translation().y();
  h = ::cartographer::transform::GetYaw(position.rotation());

// Range Data
coords.resize(trajectory_data->range_data_in_local.returns.size()*2);
  unsigned int idx = 0;
  for (const ::cartographer::sensor::RangefinderPoint &point :
    trajectory_data->range_data_in_local.returns) {
    coords[idx]   = point.position(0);
    coords[idx+1] = point.position(1);
    idx += 2;
  }
```

Is there something else I should be doing to have the LocalSlamResultCallback() RangeData position correctly reflect the initial trajectory position?

closed 2020-03-31 by glucas350

shreyasgokhale, 2020-09-20:
Hi @glucas350 , I am facing a similar issue. If you don't mind, can you please share how did you solve this?

glucas350, 2020-09-21:
I realized that I wasn't transforming the range_data_in_local into the global frame.  

`
::cartographer::transform::Rigid3d globalTransform
    = mapBuilder->pose_graph()->GetLocalToGlobalTransform(trajectoryId);
`