cartographer-project/cartographer #1675: Project Status update

catskul, 2020-03-04:
Last commits were in May 2019 by @gaschler @jkammerl 

No comments on issues or pull requests.

No response to question about next open house: https://github.com/googlecartographer/cartographer/issues/1622

No response to question about release: https://github.com/googlecartographer/cartographer/issues/1594

@wohe @SirVer What's the status of this project? Has it been dropped by the team?

andre-nguyen, 2020-03-09:
I heard from @SteveMacenski (who himself heard from other people) that the project has indeed been canned.

Kind of a shame really.

SteveMacenski, 2020-03-09:
Well, no one abandons a project (or an entire cloud robotics platform) because things were working super well. 

Can’t speak to the official nature of the project status, but certainly have heard enough negative things about it from Google employees to know its probably not going to be resurrected.

andre-nguyen, 2020-03-09:
Damn, would be great to hear from the old team or have some kind of post-mortem blog post up so the community can get a glimpse at lessons learned from this project.

hadiTab, 2020-03-31:
Are there any other similar open-source projects you guys are familair with?
Hadn't used Cartographer for a couple of years, was hoping to come back to some greate improvements right now.

Specifically, I'm looking for another 3D mapping and localization implementation that has loop closures.

andre-nguyen, 2020-03-31:
Off the top of my head there's 

- https://github.com/erik-nelson/blam 
- https://github.com/RobustFieldAutonomyLab/LeGO-LOAM
- https://github.com/koide3/hdl_graph_slam
- https://github.com/ethz-asl/segmap

MichaelGrupp, 2020-03-31:
It depends also what kind of sensor input you want. Research is more focused on camera-based SLAM since years, so you probably find more recent developments there than in LIDAR SLAM. But with any research code there are varying degrees of quality / maintenance.

One positive example is: https://github.com/MIT-SPARK/Kimera

catskul, 2020-06-16:
Looks like @wohe is bringing some things up to date! Perhaps he can chime in.

donrv, 2020-06-27:
I tried all above mentioned packages and nothing matches with Cartographer.
Mapped 15000km so far with cartographer and still going strong.

MichaelGrupp, 2020-10-17:
The project is now owned by the OSRF and maintained by volunteers - currently Wolfgang (Lyft), Bo (Enway) and myself (Magazino). With the option to involve more people that are experienced with the project and are motivated to contribute.

As you might have seen already from the commit history here and in cartographer_ros, we have been busy with maintenance tasks but are also merging fixes and new features again. There's still a lot of work to be done and things might appear a bit slow for now, but this was to be expected after a year of hibernation. We might announce more things on the way, so some things might change.

Overall I'm pretty happy that it's now run as a community-maintained project! Hope this answers some of your questions.

imstevenpmwyca, 2021-02-25:
> The project is now owned by the OSRF and maintained by volunteers - currently Wolfgang (Lyft), Bo (Enway) and myself (Magazino). With the option to involve more people that are experienced with the project and are motivated to contribute.

@MichaelGrupp Does this also include the cartographer_ros package?

MichaelGrupp, 2021-03-15:
@imstevenpmwyca Yes, also cartographer_ros.

maracuya-robotics, 2021-11-12:
Got quiet again. What's the current status?

ciwedwang, 2022-08-17:
Make cartographer great again :)