cartographer-project/cartographer #1684: Question regarding gravity aligned pose and role of IMU data in pose prediction

gsh89, 2020-04-17:
I am confused as to why the computing the gravity aligned pose is necessary and what purpose does it serve. I was looking at local trajectory builder code especially at the pose prediction part and came across [this]https://github.com/cartographer-project/cartographer/blob/bcd5486025df4f601c3977c44a5e00e9c80b4975/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L225). 

My second question pertains to the role of IMU data in pose prediction for the scan matcher. It looks like the acceleration data is used only for determining the gravity direction as mentioned in this [issue](https://github.com/cartographer-project/cartographer_ros/issues/960). How does it predict the pose then ? Also, how are the angular velocity measurements used ?

garlinplus, 2020-04-22:
用来把点云转到大地系

MichaelGrupp, 2021-03-15:
Gravity alignment is useful if your 2D scans are not always aligned (e.g. scan backpack use case).

Angular velocity is used in the pose extrapolator if available, but currently not accelerations. We use odometry (external or based on scan matching) for translation.

closed 2021-03-15 by MichaelGrupp