cartographer-project/cartographer #1689: Separation of parts when the cartographer node is running

zhongsy-zsy, 2020-05-18:
I have a question that I hope to be answered。（This is for research  @only）
)

this is my robot model in gazebo 
[image]

have tf as below
[image]
but when i start cartographer node ,the issue appeared
here is my .lua fie and launch file
```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "laser_link",
  published_frame = "laser_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.use_imu_data = false

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 6.0
TRAJECTORY_BUILDER_2D. missing_data_ray_length = 5.

return options
```
```
<launch>
  <arg name="bag_filenames" value="/home/zhongsy/2020-05-09-17-10-43.bag"/>
  <arg name="no_rviz" default="false"/>
  <param name="/use_sim_time" value="true" />

  <group unless="$(arg no_rviz)">
    <node name="rviz" pkg="rviz" type="rviz" required="true"
        args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />

    <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
        type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
  </group>

  <node name="cartographer_offline_node" pkg="cartographer_ros"
      required="$(arg no_rviz)"
      type="cartographer_offline_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basenames myrobot.lua
          -urdf_filenames $(find cartographer_ros)/urdf/cotek_with_scan.urdf
          -bag_filenames $(arg bag_filenames)"
      output="screen">
	<remap from="scan" to="/scan" />
	
  </node>
</launch>
```
then i start rviz and tf
[image]
[image]

now  I want to know why my lidar is off my robot and how to solve it

kdaun, 2020-05-18:
Hi @zhongsy-zsy 
take a look at the [Configuration Documentation](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html).
Usually, if an `odom` frame is available, the map-odom transform is published.
Try setting the published frame to odom:
`published_frame = "odom"`
and as you already provide an odom frame, set:
`provide_odom_frame = true`.

zhongsy-zsy, 2020-05-18:
@kdaun  Thank you very much for taking the time out of your busy schedule to answer my questions.
This method can solve my problem.
But I don't understand why this is the solution
Can you give me more details

i only know :
if provide_odom_frame=true the tf will become map->odom->published_frame
if provide_odom_frame=false the tf will become map->published_frame
so, I don't know what's wrong with my previous configuration

zhongsy-zsy, 2020-05-21:
In short, typically fixed_frame on Rviz is the map_frame called "map". As its name shows that map_frame is the frame of map, and it should be the root in tf tree. tracking_frame is usually the frame of IMU if IMU is available, and you've done a great job.
The result of cartographer is included in the transform from map_frame to published_frame. Cartographer will always try to maintain such a transform. For example, if your robot has an odometer, you should set published_frame to the frame of odometer and the tf tree will look like map_frame -(cartographer)-> odom_frame(published_frame) -(odometer node)-> base_link. Otherwise, you might want to choose base_link as your published_frame and your tf tree will be something more simple like map_frame -(cartographer)-> base_link(published_frame).
Finally, if provide_odom_frame is set, cartographer will publish tf transform from odom_frame to published_frame. It would be very useful when the odometer node didn't publish tf message itself.
Hope it helps and good luck.

closed 2020-05-21 by zhongsy-zsy