cartographer-project/cartographer #1695: How to choose the type of IMU to achieve certain resolution of SLAM?

xibeisiber, 2020-06-04:
Hi all,
I want to use cartographer to carry out 3D slam with resolution 10cm. The lidar I select is PandarQT 64-Channel Short-Range Mechanical LiDAR. 
But I'm not sure which IMU to choose. I searched this but didn't get much useful information. 

Can anyone recommend some types of IMU to do 3D slam with resolution 10cm? Thanks very much!

ikhann, 2021-02-09:
Xsens MTi-610 IMU
Sensonor STIM 300

handmat78, 2021-02-16:
Hello @xibeisiber ,
To achieve 3D slam with resolution 10cm resolution with your Lidar, the [Ellipse-I](https://www.sbg-systems.com/products/ellipse-micro-series/#ellipse-micro-inertial-measurement-unit-imu) from SBG Systems is a good cost-effective solution. Its performance will match your requirements, and with [ROS](https://index.ros.org/p/sbg_driver/github-SBG-Systems-sbg_ros_driver/#melodic) and [ROS2](https://index.ros.org/p/sbg_driver/github-SBG-Systems-sbg_ros2/#foxy) SBG Systems drivers Ellipse-I is also straight forward to integrate.

xibeisiber, 2021-02-16:
Thanks very much for your information~@ikhann@handmat78

closed 2021-03-15 by MichaelGrupp