cartographer-project/cartographer #1708: pose_extrapolator.cc

xbcdbc, 2020-06-23:
At line 115，i see “const transform::Rigid3d odometry_pose_delta =odometry_data_newest.pose.inverse() * odometry_data_oldest.pose;”  However，at line 178， i see
“ angular_velocity_from_poses_ =
      transform::RotationQuaternionToAngleAxisVector(
          oldest_pose.rotation().inverse() * newest_pose.rotation()) /
      queue_delta;”，I think this code should be “ angular_velocity_from_poses_ =
      transform::RotationQuaternionToAngleAxisVector(
           newest_pose.rotation().inverse() * oldest_pose.rotation()) /
      queue_delta;”？wait for your ans，thanks.

xbcdbc, 2020-06-23:
this question is for the file "pose_extrapolator.cc"

vagrant-drift, 2020-07-07:
because   first case   divide ：odometry_time_delta    const double odometry_time_delta =
      common::ToSeconds(odometry_data_oldest.time - odometry_data_newest.time);

but second case     divide ：queue_delta const double queue_delta = common::ToSeconds(newest_time - oldest_time);

开始被你带弯了