cartographer-project/cartographer #1709: Remove legacy serialization format migration code?

MichaelGrupp, 2020-06-23:
There's a bunch of code that was useful for migrating old pbstream files to the format that was established ~2 years ago with v1.0. ([serialization_format_migration](https://github.com/cartographer-project/cartographer/blob/master/cartographer/io/serialization_format_migration.h), `pbstream migrate` CLI, protos...).

We can consider if it makes sense to keep all of this or if we can clean up. I assume the amount of people who still rely on pre-1.0 pbstream files should be low.

wohe, 2020-08-03:
I looked into this and it seems to be two different things:

1. There is a legacy format and a tool `pbstream_main migrate`. My understanding is that old maps cannot be loaded before migration. I think it is safe to assume that everyone already converted all interesting maps, and if not they can use any version of Cartographer from the last year to do it. I would say it is safe to remove this now.

2. There is the histogram backwards compatibility from #1710 which seems to be fixed on the fly and _only_ there, with no migration tool available. Which means if someone is using old maps they just work right now and are migrated each time they are loaded. If we remove this support we would break people and there is no migration tool. We might still want to do it for the reasons mentioned in the PR, but we should at least be prepared to add a migration tool if needed.

@schwoere FYI, in case you have more info.

wohe, 2020-08-03:
FYI I will have a look at changing the migration tool as explained above: remove the legacy code to migrate to the new stream format, add new code to migrate to submaps with histograms.

closed 2020-08-17 by wohe