cartographer-project/cartographer #1712: Cartographer build starting failing this morning

pbaughman, 2020-06-24:
Hi,

My build of cartographer in my CI started failing this morning.  The error is

```
 CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "Findabsl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "absl", but
  CMake did not find one.
  Could not find a package configuration file provided by "absl" with any of
  the following names:
    abslConfig.cmake
    absl-config.cmake
  Add the installation prefix of "absl" to CMAKE_PREFIX_PATH or set
  "absl_DIR" to a directory containing one of the above files.  If "absl"
  provides a separate development package or SDK, be sure it has been
  installed.
-- Configuring incomplete, errors occurred!
```

It seems like it's related to this change: https://github.com/cartographer-project/cartographer/commit/ade7605facb967ae7adcbab4493eab4ead9454c6 but I don't see any notes on the PR about changes downstream users might need to make.

Can someone shed some light on what I'm expected to change to keep using Cartographer after PR #1570?

wohe, 2020-06-25:
You also need the fixes from #1711. Once you have that you need to build Abseil as it is no longer part of the cartographer build, e.g. using the provided script. Also see cartographer-project/cartographer_ros#1485.

pbaughman, 2020-06-26:
Answer is here: https://github.com/cartographer-project/cartographer_ros/pull/1485#issuecomment-649758317

closed 2020-06-26 by pbaughman

olongo, 2020-07-07:
hello, do you solve your problem, i have the same question @pbaughman