cartographer-project/cartographer #1716: New Release

jajosheni, 2020-07-09:
Hello,

Will there be a new release any time soon?
The latest release (1.0.0) is 267 commits behind the master.

jonbinney, 2020-07-14:
That might be tricky due to the abseil dependency. I think (I could be wrong) that abseil would need to get packaged either upstream for ubuntu/debian/fedora, or packaged through the OSRF repos. Either option sounds like a ton of work.

It seems that google isn't actively developing this project anymore; maybe it is worth removing the abseil dependency so that it is easier to compile and release? (I'm sure that would also be a massive amount of work though)

jajosheni, 2020-07-16:
Oh I see. Well I hope somebody who knows this repo well will have the time to create a release.

MichaelGrupp, 2020-10-17:
See https://github.com/cartographer-project/cartographer_ros/issues/1512

wolfv, 2020-11-04:
We have cartographer (and abseil-cpp) on conda-forge, and we're really interested in a fresh release: https://github.com/conda-forge/cartographer-feedstock

Conda packages work on Windows, Linux (most recent glibc based distros) & ARM64, PPC64LE and OSX.

I think another option is to statically link to abseil (not favored by distros, usually). I think google probably provides some cmake stuff to get this done?

tmartinx, 2021-02-16:
@jonbinney the abseil is finally on Debian Testing repos!

https://tracker.debian.org/pkg/abseil

;-)

jonbinney, 2021-02-16:
@tmartinx we'd need to hear from one of the maintainers of cartographer - doing releases would need push access to this repo. In general a maintainer would have to be ready to put in the time to do releases (both the initial release and later bugfix releases). That can be quite a lot of work, especially since there are multiple ros distros to potentially release for (melodic/noetic/foxy) and multiple target linux distributions to debug the build on (ubuntu/fedora/etc).

wohe, 2021-03-09:
@MichaelGrupp What needs to be done here? For the cartographer repo, I can look into adding a new tag (and update files accordingly) similar to what was done for 1.0.0 for a new version 2.0.0. Is that all we need for the cartographer side of things? I am not sure what needs to be done on the ROS side to build packages.

wohe, 2021-03-09:
@MichaelGrupp My plan would be to merge #1819, create a "2.0.0" release with a tag on then current master. Do you agree that this is moving in the right direction? FYI @bochen87

wohe, 2021-03-11:
@MichaelGrupp @bochen87 @clalancette I released a version tagged "2.0.0" of cartographer. I am not sure what the next steps regarding Cartographer ROS/ROS2 are: ideas?

Anything to do for ROS1 or should we just tag a new version there? "1.5.0" maybe?

Anything we can already do for ROS2?

MichaelGrupp, 2021-03-11:
Agree with tagging `cartographer`. The code is in a good & stable shape right now, the same applies to the `cartographer_ros` repo. Also bumping the major version makes sense since there were a lot of API changes since 2018.

We need to find a solution for abseil before creating a ROS release package though, it is not available via rosdep. Maybe there are also other small things that need to be tweaked in order to release to the ROS repos.

Since we have a few issues with discussions about a Noetic release floating around and it's a bit messy, I created one for discussion about concrete ToDos that we need to tackle now: https://github.com/cartographer-project/cartographer_ros/issues/1599

wolfv, 2021-03-11:
awesome, the PR to conda-forge is in (thanks to the bots): https://github.com/conda-forge/cartographer-feedstock/pull/13