cartographer-project/cartographer #1771: Test failures on OS X

wolfv, 2020-11-03:
Hi cartographer team!

We are packaging cartographer on conda-forge and we have a test failure on OS X.

https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=235112&view=results

https://github.com/cartographer-project/cartographer

Tobias-Fischer, 2020-11-03:
I had a quick look and the build gives away what the underlying issue is:

```
[80/333] Building CXX object CMakeFiles/cartographer.mapping.internal.2d.overlapping_submaps_trimmer_2d_test.dir/cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d_test.cc.o
In file included from ../cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d_test.cc:17:
In file included from ../cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d.h:20:
In file included from ../cartographer/common/port.h:22:
In file included from $BUILD_PREFIX/bin/../include/c++/v1/string:506:
In file included from $BUILD_PREFIX/bin/../include/c++/v1/string_view:175:
In file included from $BUILD_PREFIX/bin/../include/c++/v1/__string:57:
In file included from $BUILD_PREFIX/bin/../include/c++/v1/algorithm:643:
$BUILD_PREFIX/bin/../include/c++/v1/memory:2262:5: warning: delete called on 'cartographer::mapping::Grid2D' that is abstract but has non-virtual destructor [-Wdelete-abstract-non-virtual-dtor]
    delete __ptr;
    ^
$BUILD_PREFIX/bin/../include/c++/v1/memory:2517:7: note: in instantiation of member function 'std::__1::default_delete<cartographer::mapping::Grid2D>::operator()' requested here
      __ptr_.second()(__tmp);
      ^
$BUILD_PREFIX/bin/../include/c++/v1/memory:2471:19: note: in instantiation of member function 'std::__1::unique_ptr<cartographer::mapping::Grid2D, std::__1::default_delete<cartographer::mapping::Grid2D> >::reset' requested here
  ~unique_ptr() { reset(); }
                  ^
../cartographer/mapping/2d/submap_2d.h:42:7: note: in instantiation of member function 'std::__1::unique_ptr<cartographer::mapping::Grid2D, std::__1::default_delete<cartographer::mapping::Grid2D> >::~unique_ptr' requested here
class Submap2D : public Submap {
      ^
$BUILD_PREFIX/bin/../include/c++/v1/memory:4291:26: note: in instantiation of function template specialization 'std::__1::__shared_ptr_emplace<const cartographer::mapping::Submap2D, std::__1::allocator<const cartographer::mapping::Submap2D> >::__shared_ptr_emplace<cartographer::mapping::proto::Submap2D &>' requested here
    ::new(__hold2.get()) _CntrlBlk(__a2, _VSTD::forward<_Args>(__args)...);
                         ^
../cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d_test.cc:72:15: note: in instantiation of function template specialization 'std::__1::make_shared<const cartographer::mapping::Submap2D, cartographer::mapping::proto::Submap2D &>' requested here
        {std::make_shared<const Submap2D>(submap_2d),
              ^
1 warning generated.
```

Similarly:
```
$BUILD_PREFIX/bin/../include/c++/v1/memory:2262:5: warning: delete called on 'cartographer::mapping::RangeDataInserterInterface' that is abstract but has non-virtual destructor [-Wdelete-abstract-non-virtual-dtor]
```

wolfv, 2020-11-05:
Is fixed in master.

closed 2020-11-05 by wolfv