cartographer-project/cartographer #1789: IMU input data

dewtrullac, 2020-12-08:
Hi all,

I am using a **3DM-GX5-25** AHRS/IMU and 2 **Velodyne VLP16** lidars, mounted in a similar way as in case of the Deutsches museum 3D public data. I have managed to figure out a lot (learn from READDOCS and github issues) about the Lidar and PointCloud2 msgs but I still have poor understainding of **imu** messages. 

I am using the **KumarRobotics imu_3dm_gx4** drivers. Since the 3DM-GX5-25 IMU also includes the Extended Kalman Filter at 500 Hz rate I enable the Kalman filter. However, my IMU msgs still remain the same. I then found out, that the filtered msgs are being published under another topic (/imu_filter as the FilterOutput msg type). 

If I check individual **imu** topic msg I see that it is filled with **linear acceleration** and **angular_velocity** measurements of which covariance matrices are all zero. The **orientation** quaternion is always 0 and has the covariance matrix with the first element being -1 and others 0. 
```header: 
  seq: 35552
  stamp: 
    secs: 1606133123
    nsecs:  16914899
  frame_id: "imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0706095993519
  y: 0.0323645025492
  z: -0.0257006008178
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -1.69895464022
  y: -0.371063444654
  z: -9.55261377696
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

By checking the **imu_filter** topic msg I see that there is orientation quaternion together with the filled covariance matrix.
```
header: 
  seq: 16086
  stamp: 
    secs: 1607437180
    nsecs: 351661636
  frame_id: "imu_link"
quat_status: 1
bias_status: 1
orientation: 
  x: -0.00177353795152
  y: -0.00374871050008
  z: 0.978669941425
  w: 0.205397248268
orientation_covariance: [1.4697543520014733e-06, 0.0, 0.0, 0.0, 1.4670102927993867e-06, 0.0, 0.0, 0.0, 5.634401168208569e-05]
bias: 
  x: -0.0002450463071
  y: 0.000240148568992
  z: -5.94623888901e-05
bias_covariance: [5.347048914750019e-10, 0.0, 0.0, 0.0, 5.346166287445442e-10, 0.0, 0.0, 0.0, 6.389778706150651e-10]
bias_covariance_status: 1
orientation_covariance_status: 1
```  
Finally my question: 
Should my **imu** msgs also contain the orientation information and should the covariance matrices include some actual values (different than zero)? Or is it that the **Poseextrapolator** is doing what  the IMU is doing with the EKF filtering and is in that case the IMU EKF filtering not necessary and not being used by the cartographer?

Kind regards,
Dew

johuber, 2020-12-08:
First of all you should spend some time understanding your sensors and what they do, it is not within our scope to help you with that and also not related to cartographer.

Short answer to your question:
Only linear accelerations and angular velocities are used my cartographer.

Orientation and covariance are not used.

dewtrullac, 2020-12-09:
Thank you for your answer. Its what I wanted to know.

closed 2020-12-09 by dewtrullac