cartographer-project/cartographer #1793: kPaddingPixel effect on global pose

JakeInit, 2020-12-23:
In the final map, made from merging submaps, a border is place around the map. The value "kPaddingPixel" determines the size of that border. If a pixel resolution of 0.05m is used, does this mean that an extra 0.25m is added around the map? I'm trying to figure out if it has any affect on the global pose in the map or is it just for a more visually pleasing map?

wohe, 2021-01-08:
The `kPaddingPixel` just enlarges the computed image when visualizing (sub)maps to ensure the full data fits. It has no effect on the data used to determine any poses during SLAM as this visualization code is not used there, so global/local poses reported by SLAM are not effected.

The visualized maps are enlarged the map data structures inside Cartographer are not influenced by this.

Thus, it does of course have an effect on the origin of the (sub)maps visualized this way, so if you refer to this as local pose, it has the effect of moving the origin.

closed 2021-01-08 by wohe

JakeInit, 2021-01-08:
Thank you. This clear up  the question for me.