cartographer-project/cartographer #1794: Generating a map from a given ROS bag.

VigneshS1993, 2020-12-25:
I am working on implementing cartographer in fleet robots. Before this implementation there are some precursors which needs to be performed like computation time of the cartographer and comparing the maps with different other SLAM algorithms. Although I have gone through the cartographer documentation, I am missing something like this information of map generation using a ros bag. Please bear with my inability and I really request someone to guide me in this. Being a newbie I do understand that I need to explore a bit, but would really appreciate some help from the seniors in this case and regard. 

The problem I am facing is that I am able to bringup a launch file which spawns rviz and the robot motion with transformations being running and also the laser scan running as the bag file plays, but the mapserver is not able to receive any map during this process.

VigneshS1993, 2020-12-28:
These are my .lua and .launch files:

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  publish_to_tf = true,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 100,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 10

return options

<launch>
  <param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename turtlebotBag.lua"
      output="screen">
  </node>      
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/my_robot_2d.rviz" />
      
  <node name="cartographer_occupancy_grid_node"
        pkg="cartographer_ros"
        type="cartographer_occupancy_grid_node"
        args="-resolution 0.05" />
  <node name="playbag" pkg="rosbag" type="play"
        args="--clock $(arg bag_filename)" />
</launch>

VigneshS1993, 2020-12-28:
As I understand from the theories that cartographer needs tf2 type, I also created a python script which converts from tf type to tf2.

Please find both the files along with the python script

VigneshS1993, 2020-12-28:
I understand I have just started understanding cartographer, I would really appreciate learning from others in this regard. Kindly support for the above issue.

VigneshS1993, 2020-12-28:
The issue I get in the terminal if I run the launch file is the following 

[ INFO] [1609169862.874061051]: I1228 16:37:42.000000 22488 map_builder_bridge.cc:135] Added trajectory with ID '0'.
[ WARN] [1609169863.149658712, 1608636878.504291405]: W1228 16:37:43.000000 22488 tf_bridge.cc:52] "base_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1609169863.350771192, 1608636878.705579498]: W1228 16:37:43.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169863.552591996, 1608636878.907061544]: W1228 16:37:43.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169863.753522532, 1608636879.108673466]: W1228 16:37:43.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169863.954873199, 1608636879.310151563]: W1228 16:37:43.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169864.156841155, 1608636879.511825357]: W1228 16:37:44.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169864.358527352, 1608636879.713340389]: W1228 16:37:44.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169864.560199777, 1608636879.914908677]: W1228 16:37:44.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169864.761614818, 1608636880.116452663]: W1228 16:37:44.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169864.964362887, 1608636880.319258475]: W1228 16:37:44.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169865.166062980, 1608636880.520814172]: W1228 16:37:45.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169865.367192323, 1608636880.722370026]: W1228 16:37:45.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169865.569722850, 1608636880.923898616]: W1228 16:37:45.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169865.770045877, 1608636881.125469022]: W1228 16:37:45.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169865.971732484, 1608636881.327003676]: W1228 16:37:45.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169866.173794378, 1608636881.528634648]: W1228 16:37:46.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169866.376006236, 1608636881.730319325]: W1228 16:37:46.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169866.576829845, 1608636881.931747720]: W1228 16:37:46.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169866.778289655, 1608636882.133180757]: W1228 16:37:46.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169866.980470817, 1608636882.334601558]: W1228 16:37:46.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169867.181234238, 1608636882.536268247]: W1228 16:37:47.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169867.383117692, 1608636882.737820449]: W1228 16:37:47.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169867.584533767, 1608636882.939489059]: W1228 16:37:47.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169867.785984249, 1608636883.141381033]: W1228 16:37:47.000000 22488 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169867.987485372, 1608636883.342873412]: W1228 16:37:47.000000 22488 tf_bridge.cc:52] Could not find a connection between 'base_link' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.
[ WARN] [1609169868.189823876, 1608636883.544384504]: W1228 16:37:48.000000 22488 tf_bridge.cc:52] Could not find a connection between 'base_link' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.
[ WARN] [1609169868.390850366, 1608636883.745916437]: W1228 16:37:48.000000 22488 tf_bridge.cc:52] Could not find a connection between 'base_link' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.
[ WARN] [1609169868.592609914, 1608636883.947482382]: W1228 16:37:48.000000 22488 tf_bridge.cc:52] Could not find a connection between 'base_link' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.
[... cut ...]
[ WARN] [1609169870.810132131, 1608636886.164550326]: W1228 16:37:50.000000 22488 tf_bridge.cc:52] "base_scan" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169871.011187457, 1608636886.366108923]: W1228 16:37:51.000000 22488 tf_bridge.cc:52] "base_scan" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1609169871.213338712, 1608636886.567873564]: W1228 16:37:51.000000 22488 tf_bridge.cc:52] "base_scan" passed to lookupTransform argument source_frame does not exist. 

And the rviz screen
[image]
 looks like

mohamedsayed18, 2021-01-14:
To fix this warning I think you need to rename the frames in your URDF file

VigneshS1993, 2021-01-14:
@mohamedsayed18 this is my URDF file 
can you direct be to the right spot ?
  <link name="imu_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

  <link name="horizontal_laser_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="vertical_laser_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link" />

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu_link" />
    <origin xyz="0 0 0" />
  </joint>

  <joint name="horizontal_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="horizontal_laser_link" />
    <origin xyz="0.1007 0 0.0558" />
  </joint>

  <joint name="vertical_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="vertical_laser_link" />
    <origin rpy="0 -1.570796 3.141593" xyz="0.1007 0 0.1814" />
  </joint>

mohamedsayed18, 2021-01-15:
I think you didn't attach the URDF I can't see it

VigneshS1993, 2021-01-15:
@mohamedsayed18  Apologies 
Please find the attached file in this link below

https://github.com/VigneshS1993/cartographerIssue/blob/main/backpack_2d.urdf

mohamedsayed18, 2021-01-15:
I think your URDF is good but I can't see the base_footprint link
general tips: 
make sure your tf are consistent check with `rosrun tf2 view_frames`
check out this [tutorial](https://ouster.com/blog/building-maps-using-google-cartographer-and-the-os1-lidar-sensor/) it is great and may help you.

VigneshS1993, 2021-01-26:
> I think your URDF is good but I can't see the base_footprint link
> general tips:
> make sure your tf are consistent check with `rosrun tf2 view_frames`
> check out this [tutorial](https://ouster.com/blog/building-maps-using-google-cartographer-and-the-os1-lidar-sensor/) it is great and may help you.

Thank you Mohamad

closed 2021-01-26 by VigneshS1993