cartographer-project/cartographer #1797: Using cartographer without ROS

Juliocasta5, 2021-01-08:
HI,
I've been using Cartographer in ros melodic during some weeks, but I really need to use it without Ros.
If someone could guide me or give me some tips to do it, I'll be very pleased.

Thanks.

MichaelGrupp, 2021-01-11:
Related: https://github.com/cartographer-project/cartographer/issues/1229

From a library perspective, the `MapBuilder` is the central entity for SLAM in cartographer. Looking at what cartographer_ros does internally (e.g. map_builder_bridge) surely helps even if you don't want to use ROS yourself.

JakeInit, 2021-01-18:
As Michael said, "MapBuilder" is the center piece of running cartograppher. If you look in cartographer/mapping, there is a file called map_builder_test.cc. There are multiple scenarios, both 2d and 3d that can help show how to run cartographer without ros. This test creates dummy data, that you would just replace with either bag data or real data. Start with using lidar data only to begin with. It is easy and good enough for getting your first maps. Look at how they import their configurations so you can have your own file to edit when changing cartographer parameters. After importing parameters, set up your trajectory. Just use one trajectory to start (I have always only needed one so far) and hold onto that trajectory id. You will use it to insert the data you get from lidar. Once you start inserting data, you can start pulling local poses and then use the local to global transform to get your global poses. Also, go ahead and use the slam call back they create, or you can look at what ROS does.

closed 2021-03-15 by MichaelGrupp