cartographer-project/cartographer #1801: long and similiar corridor problem

PeakDing, 2021-01-13:
when I use the 2d laser data and the corridor is long and similiar, I found the ceres_scan_matcher_ or  real_time_correlative_scan_matcher_ would get the wrong pose_observation, can you give me some advices?

johuber, 2021-01-13:
This is a general limitation of SLAM (in particular 2D).

Nevertheless, two things you can try if feasible:

* Use wheel odometry and give it a high weight in the optimization
* Increase the sensor range that's considered in the scan matcher, such that "start" or "end" of the corridor are always within the horizon

PeakDing, 2021-01-13:
to a certain extent, the old version which used UKF is probably going to deal it，and I try to use reflector information to let front-end more rubust，but that the front-end more like a back-end , use VINS-slam strategy？

closed 2021-03-15 by MichaelGrupp