cartographer-project/cartographer #1803: Not possible to use IMU based extrapolator

kfabian, 2021-01-26:
Hi all,
I tried to use the IMU based extrapolator. But I get a "Not implemented!" error. It looks like this error still gets thrown if if you set the option to true, even so the implementation was added in the mean time. Is this the indented behavior?

https://github.com/cartographer-project/cartographer/blob/018dded2f9a0adb7e77003d05662ada251b541f5/cartographer/mapping/pose_extrapolator_interface.cc#L87-L92

Thanks

MichaelGrupp, 2021-01-26:
@wohe @schwoere From a quick look, this looks like an unnoticed bug in PR #1732 to me because the extrapolator is implemented but it's unreachable code. Can you clarify this?

wohe, 2021-01-26:
@kfabian @MichaelGrupp You are right, this is not the intended behavior. I sent a PR to fix it. Note though, that the IMU based extrapolator probably needs some testing/tuning to make it useful.

closed 2021-01-26 by wohe

schwoere, 2021-01-28:
@kfabian As indicated by @wohe, I believe the plan was to remove the `CHECK` once good parameters for the official datasets were found. Anyway, the implementation itself is complete and was tested with the 3D version of cartographer, so we can remove the `CHECK` (thanks @wohe for fixing).  Also see `TODO` in #1732 (https://github.com/cartographer-project/cartographer/pull/1732/files#diff-8d181dbe7716a2ccbf024b6eebc0c8c2ed0bfc3cbe18a2fc151dcdd3ae21195bR72) for the parameter tuning.

PR #1804 (2021-01-26): Remove CHECK preventing the use of the IMU-based extrapolator.
Fixes #1803.
commit a48f12df (2021-01-26): Remove CHECK preventing the use of the IMU-based extrapolator. (#1804)

Fixes #1803.