cartographer-project/cartographer #1807: Drawing map from pbstream

lelongg, 2021-02-15:
I need to draw a map from a pbstream produced by cartographer. I have the submaps pose from the trajectory field and the local_pose from the serialized Submap, however I can't make sense of any of these transformations but the submaps pose never match. I already tried a lot of combinations but the only source of information I found (https://github.com/cartographer-project/cartographer/issues/1507) is lacking.

The goal is to be able to display a pbstream in a third party tool, so using the available conversion nodes is not an option unfortunately.

In short: how am I supposed to combine data from a pbstream to produce a map ?

johuber, 2021-02-16:
If you look at [this](https://github.com/cartographer-project/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/assets_writer_main.cc) that may help. Among some other things it creates a png image of the map from the pbstream.

I assume what you are trying to do is fairly similar and I hope this helps then.

lelongg, 2021-02-16:
From what I understand, this process is used to aggregate, filter and output sensor data saved in a rosbag according to the localization data stored in a pbstream.

What I am trying to achieve is merging the probability grids already stored in the pbstream.

MichaelGrupp, 2021-02-17:
I would recommend to take a look which API functions are used in here: https://github.com/cartographer-project/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/pbstream_to_ros_map_main.cc

This tool reads a pbstream and creates an image from it. You can ignore the part where it writes the PGM and YAML files for ROS, but the creation of an `cartographer::io::image` from the pbstream is only a few lines of code. You could base your code on this tool and use the image for your purposes with custom code.

lelongg, 2021-02-17:
I didn't mention it but I'm writing the tool in rust and I would like to avoid binding.
I skimmed through this code several times but a fresh look may have given me what I was missing.

```c++
const double max_x = limits().max().x() - resolution * offset.y();
const double max_y = limits().max().y() - resolution * offset.x();
*texture->mutable_slice_pose() = transform::ToProto(
    local_pose.inverse() *
    transform::Rigid3d::Translation(Eigen::Vector3d(max_x, max_y, 0.)));
```

This transformation between local_pose and slice_pose might be the piece I need.