cartographer-project/cartographer #1825: Problems with 3D local SLAM

halops, 2021-03-30:
Hi, I am having problems with the 3D local SLAM tuning. I am using 2x VLP-16 and a Microstrain 3dm-gx5-45. The main issue that I have is that the local trajectory slightly oscillates, as you can see in the picture below. My configuration is the following:

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "microstrain_imu_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 2,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

--TRAJECTORY BUILDER PARAMS
--Providing full 360 scans with two lidars (and per-point stamp)
TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 6

--Setting up a voxel filter to reduce the density of the data
VOXEL_SIZE = 0.05
TRAJECTORY_BUILDER_3D.voxel_filter_size = VOXEL_SIZE

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 12
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 200
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 20
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

Changing the num_accumulated_range_data from 2 to 6 helped reduce the oscillation, but it is still present. Thanks for your help!

[image]

tyhowell, 2021-04-14:
Sounds like a tuning question - if you are using the ROS velodyne driver have you tried setting _npackets_ to 1? That would provide ~75 packets per rotation to Cartographer, so you would need to increase _TRAJECTORY_BUILDER_3D.num_accumulated_range_data_ appropriately, the original authors had it set to 160 for two velodynes.

halops, 2021-04-14:
I am working offline with recorded data and I can't test this right now, I will try this as soon as I can. However, I fail to understand how should this produce a different effect to the mapping, since it should be equivalent to providing full 360º scans with per-point timestamp, right?
Thanks for your help!