cartographer-project/cartographer #1826: overlapping trimmer crashed after delete frozen trajectory

shamengjun, 2021-04-15:
Hi there, I've tried this DeleteTrajectory service to try to remove a loaded frozen trajectory. It works well and the frozen trajectory is deleted. However, the code crashes after the frozen trajectory is deleted. The error is shown as below:

terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at

I find it happens in constraint_builder_2d.cc. The line of "(*callback)(result);" I used the latest master for cartographer which has all the updates above. Does anyone have any idea why it happens and how to resolve it? Thanks very much in advance.

_Originally posted by @cuicuihaohaoxuexi in https://github.com/cartographer-project/cartographer/issues/1767#issuecomment-806189179_

shamengjun, 2021-04-15:
When in cartographer localization mode, I load one the pbstream(trejectory 0)as false load_frozen_state(default true), In my case, once when overlapping trimmer detect one trajectory 0 's submap need to be trimmed, after pose graph trim it, then the program crash, 
 error is:
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
but when load pbstream as true load_frozen_state, and trim newest trajectory's submap that doesn't occur, how can I solve it? Can anybody give me some idea?

shamengjun, 2021-04-15:
When in cartographer localization mode, I load one the pbstream(trejectory 0)as false load_frozen_state(default true), In my case, once when overlapping trimmer detect one trajectory 0 's submap need to be trimmed, after pose graph trim it, then the program crash,
error is:
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
but when load pbstream as true load_frozen_state, and trim newest trajectory's submap that doesn't occur, how can I solve it? Can anybody give me some idea?
the ternimal log is here
[crash_log.txt](https://github.com/cartographer-project/cartographer/files/6316292/crash_log.txt)

shamengjun, 2021-04-15:
When in cartographer localization mode, I load one the pbstream(trejectory 0)as false load_frozen_state(default true), In my case, once when overlapping trimmer detect one trajectory 0 's submap need to be trimmed, after pose graph trim it, then the program crash,
error is:
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
but when load pbstream as true load_frozen_state, and trim newest trajectory's submap that doesn't occur, how can I solve it? Can anybody give me some idea?
the ternimal log is here
[crash_log.txt](https://github.com/cartographer-project/cartographer/files/6316304/crash_log.txt)

shamengjun, 2021-04-15:
After my debug, Maybe at this is occur error, but I don't know the reason and don't know how to fix it!
the function call process is:
constraint_builder_2d.cc -->RunWhenDoneCallback() ---> (*callback)(result)
pose_graph_2d.cc --->HandleWorkQueue() --->RunOptimization()--->const auto local_to_old_global = ComputeLocalToGlobalTransform(data_.global_submap_poses_2d, trajectory_id)--->return transform::Embed3D(
             global_submap_poses.at(last_optimized_submap_id).global_pose) *
         data_.submap_data.at(last_optimized_submap_id)
             .submap->local_pose()
             .inverse();

shamengjun, 2021-04-15:
After my debug, Maybe at this occur error, but I don't know the reason and don't know how to fix it!
the function call process is:
constraint_builder_2d.cc -->RunWhenDoneCallback() ---> (*callback)(result)
pose_graph_2d.cc --->HandleWorkQueue() --->RunOptimization()--->const auto local_to_old_global = ComputeLocalToGlobalTransform(data_.global_submap_poses_2d, trajectory_id)--->return transform::Embed3D(
global_submap_poses.at(last_optimized_submap_id).global_pose) *
data_.submap_data.at(last_optimized_submap_id)
.submap->local_pose()
.inverse();

shamengjun, 2021-04-15:
After my debug, Maybe at this occurs error, but I don't know the reason and know how to fix it!
the function call process is:
constraint_builder_2d.cc -->RunWhenDoneCallback() ---> (*callback)(result)
pose_graph_2d.cc --->HandleWorkQueue() --->RunOptimization()--->const auto local_to_old_global = ComputeLocalToGlobalTransform(data_.global_submap_poses_2d, trajectory_id)--->return transform::Embed3D(
global_submap_poses.at(last_optimized_submap_id).global_pose) *
data_.submap_data.at(last_optimized_submap_id)
.submap->local_pose()
.inverse();

GrantZheng86, 2021-07-09:
I am having the same exception. I did not use delete trajectory and only ran cartographer with the overlapping trimmer. 

Cartographer did not give me any errors when running in localization only mode, i.e. the "overlapping_submaps_trimmer" option is disabled in my lua file. Localization worked fine for both the original map (only trajectory 0) and the updated map (two trajectories).

I am now suspecting that the error is caused by the "overlapping_submaps_trimmer_2d.cc" file, but I could not find the calling of map::at() anywhere in this .cc file. And there are two places in the "constraint_builder_2d.cc" where the map::at() is called. My theory so far is that the trimmer has already trimmed the old submaps, but the constraint builder is still looking for the non-esist submaps. I am still working on testing this theory.

mksetaro, 2021-09-26:
Hi @GrantZheng86 and @shamengjun ,

I've just opened #1859 regarding the exception you're having, take a look and I hope it can help you.

Cheers,
Michelangelo.