cartographer-project/cartographer #1829: Command '['ninja', '-j12', '-l12']' returned non-zero exit status 1 on compile

tomkimsour, 2021-04-21:
I did everything listed on this page https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation .
But once i run `catkin_make_isolated --install --use-ninja` i get this error :

`[263/382] Building CXX object CMakeFiles/cartographer.mapping.inter...rapher/mapping/internal/constraints/constraint_builder_3d_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['ninja', '-j12', '-l12']' returned non-zero exit status 1`
  
  I tried deleting the package cartographer then cloning it but nothing changed. 
  My current ditro is Ubuntu 16.04 and ros version is kinetic.

tomkimsour, 2021-04-22:
I tried compiling the repository manualy with : 
`mkdir build && cd build`
`cmake ..`
`sudo make install`

and i get errors on the library google::protobuf

`undefined reference to `google::protobuf::internal::InitProtobufDefaults()'`

On this  [page](https://stackoverflow.com/questions/30124264/undefined-reference-to-googleprotobufinternalempty-string-abicxx11) they say it might be due to gcc version but the **gcc version i have is 5.5.0**, am I doing this correctly. I don't understand very well what i am supposed to do here.

tomkimsour, 2021-04-22:
Ok, I'm closing this, I did not try this solution but downgrading protobuf version might solve it, latest version today is := 3.15.18. Maybe putting it to 3.0.0 would work.

closed 2021-04-22 by tomkimsour

Fxx0105, 2024-04-08:
I made the same mistake as you. Did you finally solve the problem?

Fxx0105, 2024-04-08:
I am troubled by this problem now, and I hope you can help me

tomkimsour, 2024-04-10:
This is getting quite old now but I don't think I have solved it unfortunatly.