cartographer-project/cartographer #1830: The x and y direction is changed in MapLimits::GetCellIndex, why must do that? And the max_ is meaning for what?

kyo055, 2021-04-27:


sharknow, 2023-03-13:
In cartographer/mapping/2d/map_limits.h
Eigen::Array2i GetCellIndex(const Eigen::Vector2f& point) const {
// Index values are row major and the top left has Eigen::Array2i::Zero()
// and contains (centered_max_x, centered_max_y). We need to flip and
// rotate.
return Eigen::Array2i(
common::RoundToInt((max_.y() - point.y()) / resolution_ - 0.5),
common::RoundToInt((max_.x() - point.x()) / resolution_ - 0.5));
}
I think this should be:
return Eigen::Array2i(
common::RoundToInt((max_.x() - point.x()) / resolution_ - 0.5),
common::RoundToInt((max_.y() - point.y()) / resolution_ - 0.5));

I look forward to your reply!

kyo055, 2023-05-12:
@sharknow I
 think the local coordinate is in right hand, but the grid map stores as image format which is left hand. And the x direction is always the car direction, so cartographer tries to decipt the pro
[image]
cess as the follows.

closed 2023-05-15 by kyo055