cartographer-project/cartographer #1836: catkin_make_isolated --install --use-ninja error

Aadi0902, 2021-05-09:
Hi,
I am having the following error when running `catkin_make_isolated --install --use-ninja':

Failed to process package 'cartographer':
Command '['ninja', '-j8','-l8']' returned non-zero exit status 1

I was following this documentation:
https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html

I am running Ubunut 16.04 with ROS kinetic, gcc version: 5.4.2
lua version 5.2

I tried deleting the package and reinstalling, but that did not help

georgflick, 2021-05-11:
Hi @Aadi0902 ,
do you get an error before the failed ninja command? exit status 1 could mean a lot of things.

E.g if you get an error like:
```
FAILED: docs/CMakeFiles/build_doc 
from sphinx.cmd.build import main
ModuleNotFoundError: No module named 'sphinx'
```
you just need to install sphinx with `pip3 install sphinx ` and it should work.

Aadi0902, 2021-05-11:
@georgflick Here is what I get:
[image]

georgflick, 2021-05-12:
@Aadi0902 seems like a compiler issue to me.
I just compiled again and it worked fine with:
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
```