cartographer-project/cartographer #1839: laser ray trace maybe bug

shamengjun, 2021-05-14:
Hello,everyone:
    When I use cartographer mapping, there  maybe laser ray trace occur some stranges, as follows piture:
[image]
 Why the points behind of the box is hit and the yellow part is mapped while the other box is normal?  But use gmapping this is normal and correct, the result is:
[image]
Can anyone give one explanation ? and how can I fix this bug in cartographer?

shamengjun, 2021-05-14:
my configure about laser insert is:
TRAJECTORY_BUILDER_2D.min_range = 0.5,
TRAJECTORY_BUILDER_2D.max_range = 20,
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 10.
and I don't understand this and my problem diffenrences?

ibuterin, 2021-05-14:
Hi, this is an open issue.
It is explained in https://github.com/cartographer-project/cartographer/issues/947#issuecomment-519870583

shamengjun, 2021-05-14:
@ibuterin thanks, that is to say this is one cartographer self bug.

MichaelGrupp, 2021-05-14:
We are working on a solution for this. Yes, it's a bug.