cartographer-project/cartographer #1840: ordered_multi_queue.cc:45] Check failed: entry.second.finished

JakeInit, 2021-05-18:
When running cartographer, I can map and localize without any issues in my 2d environment. The issue happens when trying to delete my pointer to MapBuilderInterface.

std::unique_ptr<cartographer::mapping::MapBuilderInterface> map_builder_;
map_builder_ = std::make_unique<mapping::MapBuilder>(map_builder_options_); // options being my configurations from the lua

And for the trajectory builder
trajectory_id = map_builder_->AddTrajectoryBuilder(
        {kRangeSensorId, kIMUSensorId, kOdomSensorId}, trajectory_builder_options_,
        GetLocalSlamResultCallback());

Everything during running is fine as stated above.
Later when deconstructing my cartographerInterface class, I do the following:

```
// Finish trajectory
for (const auto &entry : map_builder_->pose_graph()->GetTrajectoryStates()) {
  if (entry.second == mapping::PoseGraph::TrajectoryState::ACTIVE) {
    auto trajectory_id = entry.first;
    map_builder_->FinishTrajectory(trajectory_id);
    while (!map_builder_->pose_graph()->IsTrajectoryFinished(trajectory_id)) {  // wait for each trajectory to finish
      std::this_thread::sleep_for(std::chrono::milliseconds(10));
    }
  }
}

// Run final optimizations
map_builder_->pose_graph()->RunFinalOptimization();
std::this_thread::sleep_for(std::chrono::milliseconds(150));

if(trajectory_builder != nullptr) {
  trajectory_builder = nullptr;
}

if(map_builder_ != nullptr) {
  map_builder_.reset();  // this is where the issue occurrs
}
```

As soon as I try to reset map_builder_, I get the following log error and then crash of the application
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0518 10:16:17.584446 15512 ordered_multi_queue.cc:45] Check failed: entry.second.finished 
*** Check failure stack trace: ***
Aborted

This is called in the trajectory collator. I would have assumed since I called finish trajectory, that all trajecories (should only be 1) are labeled as finished. This seems to consistently happen if I stop sending sensor data to cartographer for a period of time and then start sending data again before trying to close my application.

Is there a way to handle this through the MapBuilderInterface and catch it to avoid crashing the application? Or am I missing a step when shutting down cartographer?

JakeInit, 2021-05-28:
This seems to occur when trying to map an area, but sensor measurements are not pushed to Cartographer. Is there any way to prevent the crash in this scenario?

b-lala, 2022-09-28:
I have the same problem, did you solve it？

VegetarianLion258, 2023-07-06:
maybe caused by building a new trajectory but the former trajectory not be destructed. only one trajectory is allowed.