cartographer-project/cartographer #1843: can cartographer work with airsim?

jaredsjohansen, 2021-06-22:
I am running Airsim with a single drone (which has a lidar and imu).
I am also running [airsim-to-ros]( https://microsoft.github.io/AirSim/airsim_ros_pkgs.html).
With only these running, I get the following tf tree: [frames.pdf](https://github.com/cartographer-project/cartographer/files/6690395/frames.pdf).

I understand that I want my tf tree to look like this: map -> odom -> base_link.

URDF: Because there are already tf relationships between drone1, odom_local_ned, and Lidar, my urdf file is empty.  (I believe I don't need to define them.  Please correct me if I am wrong.)

LUA: I have created a my_robot.lua file that looks like this: 
```options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "drone1",
  published_frame = "odom_local_ned",
  odom_frame = "not-used",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}```

When I run cartographer, I get this error.
`[ WARN] [1624326678.959343053]: W0621 18:51:18.000000 71181 tf_bridge.cc:52] "odom_local_ned" passed to lookupTransform argument source_frame does not exist. `

I can't quite determine whether airsim is producing a tf tree that breaks the map->odom->base_link paradigm...or if I've messed something up in lua file.  I've read the [docs](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) and tried different things, but can't quite determine whether I have something amiss...or whether there is something amiss with airsim.

jaredsjohansen, 2021-06-24:
I have made some discoveries, but am still unable to get things working properly.  Here's the "closest" I've gotten things to work.

I launch airsim and airsim-to-ros.  Airsim is producing the following topics:
```/airsim_node/base_link/altimeter/barometer
/airsim_node/base_link/environment
/airsim_node/base_link/global_gps
/airsim_node/base_link/gps/gps
/airsim_node/base_link/imu/Imu
/airsim_node/base_link/imu/imu
/airsim_node/base_link/lidar/LidarCustom
/airsim_node/base_link/magnetometer/magnetometer
/airsim_node/base_link/odom_local_ned
/airsim_node/base_link/vel_cmd_body_frame
/airsim_node/base_link/vel_cmd_world_frame
/airsim_node/gimbal_angle_euler_cmd
/airsim_node/gimbal_angle_quat_cmd
/airsim_node/origin_geo_point
/rosout
/rosout_agg
/tf
/tf_static
```

One of the topics is `/tf`.  The [tf_tree](https://github.com/cartographer-project/cartographer/files/6705742/tf_tree.pdf) has some extraneous stuff and the` odom->baselink` transformation does not appear to be correct.  (Note that there is a settings.json file in airsim where you can specify the name of the drone.  Its default is "drone_0".  I changed the name to "base_link".  That is why the topics are `/airsim_node/**base_link**/odom_local_ned` instead of `/airsim_node/**drone_0**/odom_local_ned` and also why the tf tree has `world_ned->**base_link**->base_link/odom_local_ned` instead of` world_ned->base_link->**drone_0**/odom_local_ned`.)

I modify the airsim-to-ros launch file to remap the `/tf` topic to `/garbage` and `/tf_static` to `/garbage2`.  After relaunching airsim-to-ros, there is no tf tree being produced on the `/tf` topic.

This means I can now configure cartographer to produce the tf tree, including the odom frame.  My lua file looks like this:
```include "map_builder.lua"
include "trajectory_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.2)
POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.65
POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
return options
```

When I run cartographer, the tf tree looks like [this](https://github.com/cartographer-project/cartographer/files/6705780/tf_tree_cartographer.pdf) (which is simple and appears to be correct).

I then command the drone to fly around the map (using airsim commands).  As the drone is flying around, I can visualize the laserscan points in rviz and see them following the path of the drone.  The map that is produced by cartographer looks pretty good: [map](https://user-images.githubusercontent.com/84985880/123192288-6f54be00-d457-11eb-869d-6219fa962131.jpg).

There are a few problems though.  If I visualize the "baselink frame" in rviz, it stays pretty close to the origin.  It moves maybe 2 meters in a horizontal-ish direction (while the commands I gave the robot moved it all around the building).   If I run move_base, the local costmap stays fixed around the (almost non-moving) base_link frame rather than following the laserscan points all throughout the building.  The laserscan points appear to be moving according to the locations produced by `/airsim_node/drone1/odom_local_ned`.

I figured `/airsim_node/drone1/odom_local_ned` was important odometry data...so I updated my airsim-to-ros launch file to remap `/airsim_node/drone1/odom_local_ned` to `/odom`.  I also modified my lua file to set `use_odometry = true`.  Now, when I rerun cartographer, no tf tree is produced and cartographer throws this warning:
`[ WARN] [1624501629.273111212]: W0623 19:27:09.000000 28463 tf_bridge.cc:52] "base_link" passed to lookupTransform argument target_frame does not exist.`

I feel that I am close to getting things working...but have been unable to get passed this last hurdle of getting base_link to move around the map in rviz in the same way that is moving around the building in airsim.  I suspect that something with the tf tree being produced may not be perfect, but am uncertain what that might be.
 
Note: During all this, my URDF file is essentially empty.  I do not have any connections to the outputs of airsim, and do not know if/how I can do that.

kellen080, 2022-04-14:
Do you find any solution now ?

jaredsjohansen, 2022-04-14:
Nope.  Just a hacky workaround that works so-so.

hafnerdMUC, 2022-08-19:
Hey @jaredsjohansen, thanks for sharing your solution! Im working on a similar project and experience also different issues with cartographer. Do you have a solution now? I cant even get a map in the way you described it above. Can you share your settings.json file?

jaredsjohansen, 2022-08-19:
No, I never found a proper solution.  I only ever used the hacky workaround.  

I'm attaching two settings.json files that I found associated with this project.  (I had to rename them to *.txt since github doesn't support attaching *.json files.)  It's been over a year since I've dealt with this code, so I don't really remember which one I ended up using.  Hopefully they provide some helpful insight to you.

[settings.txt](https://github.com/cartographer-project/cartographer/files/9382465/settings.txt)
[settings2.txt](https://github.com/cartographer-project/cartographer/files/9382466/settings2.txt)

Arrow-gh, 2023-02-15:
Hello jare, I have the same problem recently. The map is published by cartograpeher.But the car is shaking around the origin point, and the deliverd map is bad. Do you have any solution now? 
[image]

jaredsjohansen, 2023-02-15:
No, I haven't worked with AirSim for ~1.5 years.  (I had to do so for a particular customer demo, which is where this thread originated.  Now that that demo has passed, I haven't gone back to it and predominantly work in Gazebo.)

closed 2023-02-15 by jaredsjohansen