cartographer-project/cartographer #1859: Delete trajectory makes cartographer crash

mksetaro, 2021-09-26:
Hi there, 

At Yape we have been using cartographer 3d since the beginning of the project and recently we have started working on some customization to make it more suitable for our use case.

While working on that we have noticed that deleting a trajectory was causing the cartographer to crash with two different error traces.

The first one gives the following trace (already mentioned in #1826 and #1767 comment ):

`[ INFO] [1632231593.656546094]: I0921 15:39:53.000000 12803 pose_graph_3d.cc:534] Remaining work items in queue: 0
callback result pre
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at`

The second gives:

`F0921 15:25:05.650257 11545 optimization_problem_3d.cc:371] Check failed: imu_data_.HasTrajectory(trajectory_id)
[FATAL] [1632230705.650795345]: F0921 15:25:05.000000 11545 optimization_problem_3d.cc:371] Check failed: imu_data_.HasTrajectory(trajectory_id)`

After some analysis and debugging we've noticed that a missing check to ignore deleted trajectories in **OptimizationProblem3D::Solve** and  **PoseGraph3D::RunOptimization()** was causing the above mentioned errors.

We have fixed the issue but I still would like to ask if the solution is correct or if we're missing something and the patch could backfire in future.

Thanks and cheers,
Michelangelo.

bufeng-12, 2024-09-27:
您好！我已收到你的邮件！谢谢！          ——曾君

qetuo105487900, 2024-09-27:
?

michelangelosetaro, 2024-09-27:
> same problem in 2d, but no one slove it.
If you read the original post, we were able to solve it. It was pretty trivial race condition. Don't recall exactly the code but it was enough to close trajectory scheduled for deletion.

qetuo105487900, 2024-09-27:
> > same problem in 2d, but no one slove it.
> > If you read the original post, we were able to solve it. It was pretty trivial race condition. Don't recall exactly the code but it was enough to close trajectory scheduled for deletion.

so, how to add "missing check to ignore deleted trajectories" in 2d ? anyone can help me ？

michelangelosetaro, 2024-12-04:
It's been a while since I changed that code, but if my initial poste (see above) was the starting point. It probably works in the same way for [optimization_problem_2d](https://github.com/cartographer-project/cartographer/blob/master/cartographer/mapping/internal/optimization/optimization_problem_2d.h)