cartographer-project/cartographer #1871: cartographer/common/make_unique.h: No such file or directory

LukeZhi, 2021-12-28:
/home/luke/Documents/code/ugv_sdk/src/cartographer_ros/cartographer_ros/cartographer_ros/assets_writer.cc:24:10: fatal error: cartographer/common/make_unique.h: No such file or directory
   24 | #include "cartographer/common/make_unique.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [cartographer_ros/cartographer_ros/CMakeFiles/cartographer_ros.dir/build.make:63: cartographer_ros/cartographer_ros/CMakeFiles/cartographer_ros.dir/cartographer_ros/assets_writer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:15785: cartographer_ros/cartographer_ros/CMakeFiles/cartographer_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

FENGZHANG123, 2022-03-16:
Hi, have you sloved it? same problem here

FENGZHANG123, 2022-03-16:
I found the solution, the version of catographer_ros is old, update to the lastest master branch, not the pulished version