cartographer-project/cartographer #1883: [cartographer_node-1] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable"

bxtbold, 2022-05-17:
Hi, I am using `cartographer_node` on ROS 2 Foxy and Ubuntu 20.04 for my project and faced with some issues that are listed below.
 - As you can see from the `rviz` screenshot at the bottom, when fixed frame is `map`, TFs for `base_footprint`, `base_link` and `lidars` are not showing. (But mapping worked as expected)
 - Also, the error is raised in the terminal where `cartographer_node` is run.

```bash
[cartographer_node-1] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[cartographer_node-1]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[cartographer_node-1] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[cartographer_node-1]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-3] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[rviz2-3]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[cartographer_node-1] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[cartographer_node-1]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-3] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[rviz2-3]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[cartographer_node-1] [INFO] [1652772445.126293334] [cartographer_ros]: I0517 16:27:25.000000 120729 local_trajectory_builder_2d.cc:114] Range data collator filling buffer.
[rviz2-3] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
[rviz2-3]          at line 259 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
```  

The error says that there's a transform is published with itself listed as its parent. However, when I launched my robot in a gazebo world, `rqt_tree` results the following graph (I am assuming that there is no problem with transformations):
[image]

Then, I executed `cartographer_node` using the following lua configurations and opened the `rviz2`.  

```lua
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",             -- the frame that is tracked by the SLAM algorithm
  published_frame = "odom",                      -- the childframe for publishing poses  (init: odom)
  odom_frame = "odom",
  provide_odom_frame = true,                     -- If enabled, the local, non-loop-closed, continuous pose will be published as the odom_frame in the map_frame.
  publish_frame_projected_to_2d = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 2,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,             -- Timeout in seconds to use for looking up transforms using tf2 (0.2)
  submap_publish_period_sec = 0.3,                -- Interval in seconds at which to publish the submap poses (0.3)
  pose_publish_period_sec = 5e-3,                 -- Interval in seconds at which to publish poses (5e-3 -> 200Hz)
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.min_range = 0.12              -- 2D mapping min range
TRAJECTORY_BUILDER_2D.max_range = 12.0              -- 2D mapping max range
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 3.  -- Points beyond 'max_range' will be inserted with this length as empty space.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)
POSE_GRAPH.constraint_builder.min_score = 0.65     -- Threshold for the scan match score below which a match is not considered. Low scores indicate that the scan and map do not look similar.
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7   -- Threshold below which global localizations are not trusted
-- POSE_GRAPH.optimize_every_n_nodes = 0
return options
```
[image]

closed 2022-07-06 by bxtbold

vela-io, 2023-02-02:
do you have any solution for this issue?

bufeng-12, 2023-02-02:
您好！我已收到你的邮件！谢谢！          ——曾君

bxtbold, 2023-02-04:
> do you have any solution for this issue?

Hello @vela-io.

My problem cause was the empty `frame_id`. When I check `/joint_states` topic, the topic message showed that `frame_id` in the header was empty string (same as the below). So, make sure you have published your joints properly
```
header:
  stamp:
    sec: 1575196473
    nanosec: 629014545
  frame_id: ''
```

MorningFrog, 2024-03-02:
Hello, I have met the same problem, and I solved it by setting `provide_odom_frame = false`.

MorningFrog, 2024-03-02:
Sorry, maybe it's too late to provide my solution...
I have met the same problem, and I solved it by setting `provide_odom_frame = false`.

bxtbold, 2024-03-05:
> Sorry, maybe it's too late to provide my solution...
> I have met the same problem, and I solved it by setting provide_odom_frame = false.

Thank you for contributing your solution. I hope it helps others who are having the same issue :) @MorningFrog