cartographer-project/cartographer #1886: Linking CXX executable cartographer_pbstream FAILED: cartographer_pbstream

wsp666, 2022-06-22:
when I performance `catkin_make_isolated --install --use-ninja`, It shows ` Linking CXX executable cartographer_pbstream FAILED: cartographer_pbstream`. Specifically，as follows：
```C
-- Build files have been written to: /home/wsp/catkin_ws/build_isolated/cartographer/install
ninja: no work to do.
==> ninja -j16 -l16 in '/home/wsp/catkin_ws/build_isolated/cartographer/install'
[3/132] Linking CXX executable cartographer_pbstream
FAILED: cartographer_pbstream 
: && /usr/bin/c++  -O3 -DNDEBUG  -rdynamic CMakeFiles/cartographer_pbstream.dir/cartographer/io/pbstream_main.cc.o  -o cartographer_pbstream  libcartographer.a  /usr/local/lib/libceres.a  -lglog  /usr/lib/x86_64-linux-gnu/libspqr.so  /usr/lib/x86_64-linux-gnu/libcholmod.so  /usr/lib/x86_64-linux-gnu/libamd.so  /usr/lib/x86_64-linux-gnu/libcamd.so  /usr/lib/x86_64-linux-gnu/libccolamd.so  /usr/lib/x86_64-linux-gnu/libcolamd.so  /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so  -lrt  /usr/lib/x86_64-linux-gnu/libtbb.so.2  /usr/lib/x86_64-linux-gnu/libcxsparse.so  -llapack  -lblas  -llua5.2  -lm  /usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.71.0  -lglog  /usr/lib/x86_64-linux-gnu/libgflags.so.2.2.2  -lcairo  -lprotobuf  /usr/local/lib/libabsl_leak_check.a  /usr/local/lib/libabsl_hash.a  /usr/local/lib/libabsl_bad_variant_access.a  /usr/local/lib/libabsl_city.a  /usr/local/lib/libabsl_raw_hash_set.a  /usr/local/lib/libabsl_bad_optional_access.a  /usr/local/lib/libabsl_hashtablez_sampler.a  /usr/local/lib/libabsl_str_format_internal.a  /usr/local/lib/libabsl_synchronization.a  /usr/local/lib/libabsl_stacktrace.a  /usr/local/lib/libabsl_graphcycles_internal.a  /usr/local/lib/libabsl_symbolize.a  /usr/local/lib/libabsl_malloc_internal.a  /usr/local/lib/libabsl_debugging_internal.a  /usr/local/lib/libabsl_demangle_internal.a  /usr/local/lib/libabsl_time.a  /usr/local/lib/libabsl_strings.a  /usr/local/lib/libabsl_strings_internal.a  /usr/local/lib/libabsl_throw_delegate.a  /usr/local/lib/libabsl_base.a  -lpthread  /usr/local/lib/libabsl_dynamic_annotations.a  /usr/local/lib/libabsl_spinlock_wait.a  /usr/local/lib/libabsl_int128.a  /usr/local/lib/libabsl_civil_time.a  /usr/local/lib/libabsl_time_zone.a  -lpthread && :
/usr/bin/ld: libcartographer.a(histogram.cc.o): in function `cartographer::common::Histogram::ToString[abi:cxx11](int) const':
histogram.cc:(.text+0x43c): undefined reference to `absl::strings_internal::CatPieces[abi:cxx11](std::initializer_list<std::basic_string_view<char, std::char_traits<char> > >)'
/usr/bin/ld: histogram.cc:(.text+0x95a): undefined reference to `absl::strings_internal::AppendPieces(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::initializer_list<std::basic_string_view<char, std::char_traits<char> > >)'
collect2: error: ld returned 1 exit status
[18/132] Building CXX object CMakeFile...er/transform/rigid_transform_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['ninja', '-j16', '-l16']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /home/wsp/catkin_ws/build_isolated/cartographer && ninja -j16 -l16

Command failed, exiting.
```

cjchang925, 2022-06-27:
I did not encounter this problem when I installed Cartographer. Could you post what you had done before this error happened? BTW, did you clone the newest version of Cartographer?

baduc1602, 2022-11-16:
Did you fix this problem? If yes, can you show the solution to solve this.

Omiclous, 2023-10-10:
Hi, @wsp666 @baduc1602  Did you find a solution?