cartographer-project/cartographer #1899: error: ‘AutoDiffLocalParameterization’ is not a member of ‘ceres’

wanesta, 2022-08-29:
I am following this link https://google-cartographer.readthedocs.io/en/latest/ to install cartographer ,and I got this error:
```
root@ubuntu:/home/gaosm/Downloads/cartographer/cartographer/build# ninja
[10/385] cd /home/gaosm/Downloads/cartographer/cartographer/build &...ome/gaosm/Downloads/cartographer/cartographer/build/AllFiles.cmake"
Files /home/gaosm/Downloads/cartographer/cartographer/build/AllFiles.cmake and - differ
[111/385] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc.o
FAILED: CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc.o 
/usr/bin/c++ -DGFLAGS_IS_A_DLL=0 -I/home/gaosm/Downloads/cartographer/cartographer/build -isystem /usr/local/include/eigen3 -isystem /home/gaosm/Downloads/cartographer/cartographer -isystem /usr/include/lua5.2 -O3 -DNDEBUG  -pthread -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -std=gnu++1z -MD -MT CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc.o -MF CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc.o.d -o CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc.o -c /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc: In member function ‘void cartographer::mapping::scan_matching::CeresScanMatcher3D::Match(const Vector3d&, const Rigid3d&, const std::vector<cartographer::mapping::scan_matching::PointCloudAndHybridGridsPointers>&, cartographer::transform::Rigid3d*, ceres::Solver::Summary*) const’:
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:102:42: error: ‘AutoDiffLocalParameterization’ is not a member of ‘ceres’
                 absl::make_unique<ceres::AutoDiffLocalParameterization<
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:102:42: note: suggested alternative: ‘LocalParameterization’
                 absl::make_unique<ceres::AutoDiffLocalParameterization<
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                          LocalParameterization
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:102:23: error: parse error in template argument list
                 absl::make_unique<ceres::AutoDiffLocalParameterization<
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     YawOnlyQuaternionPlus, 4, 1>>())
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:101:58: error: expected primary-expression before ‘(’ token
           ? std::unique_ptr<ceres::LocalParameterization>(
                                                          ^
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:103:51: error: expected primary-expression before ‘)’ token
                     YawOnlyQuaternionPlus, 4, 1>>())
                                                   ^
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:42: error: ‘QuaternionParameterization’ is not a member of ‘ceres’
                 absl::make_unique<ceres::QuaternionParameterization>()),
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:42: note: suggested alternative: ‘QuaternionToRotation’
                 absl::make_unique<ceres::QuaternionParameterization>()),
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                          QuaternionToRotation
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:70: error: no matching function for call to ‘make_unique<<expression error> >()’
                 absl::make_unique<ceres::QuaternionParameterization>()),
                                                                      ^
In file included from /usr/include/c++/7/memory:80:0,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/common/lua_parameter_dictionary.h:21,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.h:24,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:17:
/usr/include/c++/7/bits/unique_ptr.h:832:5: note: candidate: template<class _Tp, class ... _Args> typename std::_MakeUniq<_Tp>::__invalid_type std::make_unique(_Args&& ...) <deleted>
     make_unique(_Args&&...) = delete;
     ^~~~~~~~~~~
/usr/include/c++/7/bits/unique_ptr.h:832:5: note:   template argument deduction/substitution failed:
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:70: error: template argument 1 is invalid
                 absl::make_unique<ceres::QuaternionParameterization>()),
                                                                      ^
In file included from /usr/include/c++/7/memory:80:0,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/common/lua_parameter_dictionary.h:21,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.h:24,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:17:
/usr/include/c++/7/bits/unique_ptr.h:826:5: note: candidate: template<class _Tp> typename std::_MakeUniq<_Tp>::__array std::make_unique(std::size_t)
     make_unique(size_t __num)
     ^~~~~~~~~~~
/usr/include/c++/7/bits/unique_ptr.h:826:5: note:   template argument deduction/substitution failed:
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:70: error: template argument 1 is invalid
                 absl::make_unique<ceres::QuaternionParameterization>()),
                                                                      ^
In file included from /usr/include/c++/7/memory:80:0,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/common/lua_parameter_dictionary.h:21,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.h:24,
                 from /home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:17:
/usr/include/c++/7/bits/unique_ptr.h:820:5: note: candidate: template<class _Tp, class ... _Args> typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...)
     make_unique(_Args&&... __args)
     ^~~~~~~~~~~
/usr/include/c++/7/bits/unique_ptr.h:820:5: note:   template argument deduction/substitution failed:
/home/gaosm/Downloads/cartographer/cartographer/cartographer/mapping/internal/3d/scan_matching/ceres_scan_matcher_3d.cc:105:70: error: template argument 1 is invalid
                 absl::make_unique<ceres::QuaternionParameterization>()),
```

Hanifwj, 2022-09-22:
I'm having the same problem as you, I'm using Ubuntu 20.04 and ros noetic. Is there any solution for this problem ?

Weje26, 2022-09-28:
solution is just not Install ceres-solver lastest because have same bug, I can solve problem using ceres-solver version 1.14