cartographer-project/cartographer #1912: The scanned wall is not solid with pepperl_fuchs R2000

LaskerXCB, 2022-12-01:
Hi guys,
I had a problem. I used the radar of Pepperl_fuchs r2000, and the radar was not installed in the center of the vehicle body. 
When doing 2D slam, the wall was not solid and the border of the wall was gradually faded out. 

Has anyone encountered this problem? 
I found that if I set the radar to have no offset center in my urdf file, I would not have this problem.
I have also done some tests likes,  If the urdf of the demo "deutsches Museum" is offset by 1m, the scanned object is not completely solid.
The following is the result of sweeping out the wall. Has anyone ever encountered this problem?
[image]

LaskerXCB, 2022-12-01:
urdf: 
  <joint name="horizontal_laser_joint" type="fixed">
    <parent link="base_link" />
    <child link="laser_link" />
    <origin xyz="-0.193 0.1015 0.025" rpy="0 0 -3.1368926" />
  </joint>

manhtoanbkhn, 2023-08-03:
Same problem!