cartographer-project/cartographer #1938: gcc failed: error executing CppCompile command with Bazel@HEAD

sgowroji, 2023-10-17:
https://buildkite.com/bazel/bazel-at-head-plus-downstream/builds/3391#018b3bcb-2713-4f8b-a4bf-bc799cf8a92d
Platform : Ubuntu
Logs:
```
ERROR: /var/lib/buildkite-agent/.cache/bazel/_bazel_buildkite-agent/595e58dc908d146f3afb937d2a9250aa/external/com_github_googlecartographer_async_grpc/async_grpc/BUILD.bazel:49:11: Compiling async_grpc/event_queue_thread.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target @com_github_googlecartographer_async_grpc//async_grpc:async_grpc)
```

Steps:
```
git clone -v https://github.com/googlecartographer/cartographer.git
git reset ef00de2317dcf7895b09f18cc4d87f8b533a019b --hard
export USE_BAZEL_VERSION=2ddacab80af1dca3c04e74809827c47ec4bc8498
bazel build ...

```

CC Greenteam @fweikert

sgowroji, 2023-10-26:
Hi @wohe, Can you have a look at the above issue. Thanks!

Wyverald, 2023-11-09:
Hmm, for some reason our culprit finder pinned this on `--enable_bzlmod` being flipped to true (https://buildkite.com/bazel/culprit-finder/builds/7854#018bb242-8678-4450-ba11-32be8e342f96). The error message suggests nothing that would incriminate Bzlmod, though... will have to dig a bit more.

Wyverald, 2023-11-09:
Hmm, this is probably because the bundled protobuf in `@bazel_tools` when Bzlmod is enabled is a higher version than what Cartographer's WORKSPACE says, and that one has an incompatible API change.

Given that this repo hasn't seen a commit in over a year, I'll just go ahead and disable Cartographer from Bazel CI.

bufeng-12, 2026-01-22:
您好！我已收到你的邮件！谢谢！          ——曾君