cartographer-project/cartographer #1942: IMU sensitivity is too high

EmirEvcil, 2023-12-28:
I'm trying to integrate imu (MPU9250) into google cartographer. I publish the IMU data I obtain to the /imu topic by passing it through the Madgwick filter. When I check the sensor with ros_imu_plugin, the movement seems to be very smooth and working properly. However, when you set use_imu = true in Google Cartographer, the mapping becomes very stupid and the orientation of the base_link changes very quickly, so the sensitivity seems to be very low. Even a little movement causes a lot of rotation. What is the reason of this ? I had this problem before in ros_imu_plugin, but I fixed it by playing with the parameters in the Madgwick filter, but I don't know how to solve it in google cartograph. I would also like to point out that I did 2D mapping.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = true,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.5
TRAJECTORY_BUILDER_2D.max_range = 8.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 8.5
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.2)
-- for current lidar only 1 is good value
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.65
POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35

return options

bufeng-12, 2023-12-28:
您好！我已收到你的邮件！谢谢！          ——曾君

EmirEvcil, 2023-12-30:
header: 
  seq: 780885
  stamp: 
    secs: 1703943037
    nsecs: 214104413
  frame_id: "base_link"
orientation: 
  x: -0.8021243250537446
  y: -0.021046560940722856
  z: 0.5868600885324634
  w: -0.10839209342619172
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.244140625
  y: 0.54931640625
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.267578125
  y: -0.69921875
  z: 1.32373046875
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

imu data that i publish