cartographer-project/cartographer #1949: Exploration Error while using Cartographer for Localization

Rajatsurya, 2024-01-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: Noetic
    - Firmware version: turtlebot3_msgs/VersionInfo

7. Copy and Paste the error messages on terminal.
 Extrapolation Error: Lookup would require extrapolation 0.003984088s into the future.  Requested time 1703667865.989929199 but the latest data is at time 1703667865.985944986, when looking up transform from frame [odom] to frame [map]

[ERROR] [1703667865.990101947]: Global Frame: odom Plan Frame size 22: map

[ WARN] [1703667865.990149510]: Could not transform the global plan to the frame of the controller
[ERROR] [1703667865.990196793]: Could not get local plan
got new plan
got new plan 
got new plan 
 Extrapolation Error: Lookup would require extrapolation 0.003984088s into the future.  Requested time 1703667865.989929199 but the latest data is at time 1703667865.985944986, when looking up transform from frame [odom] to frame [map]

[ERROR] [1703667865.990101947]: Global Frame: odom Plan Frame size 22: map

[ WARN] [1703667865.990149510]: Could not transform the global plan to the frame of the controller
[ERROR] [1703667865.990196793]: Could not get local plan
got new plan
got new plan
goal reached

8. Please describe the issue in detail.
I am currently facing this above issue ,here I am using Cartographer localization instead of AMCL to localize the robots.Because of this exploration error the motion of the robots is not continuous . what should I change to eliminate this error also I am using 2 robots one is equipped with an LDS 01 sensor and the other with an LDS 02 sensor the frequency with which the exploration error occurs is much lesser in the LDS 02 sensor ,I have been stuck with this problem for quite some time would be great if I can get some assistance on this

tmli11, 2026-10-21:
I met the same issue, it was caused by a TF timestamp mismatch between Cartographer and ROS movs_base. 

1. Global Planner : it computes a global path in the map frame and stamps the path with the current time (e.g., 4631.775s).

2. Local Planner(DWA In my case): After receiving the path, it attempts to transform it from the map frame to the odom frame, because the local costmap's global_frame is odom.

3. Cartographer publishes the transform between **map and odom**. However, it will always introduce a few milliseconds of processing latency.

4. **Failure: local planner requests the map-to-odom transform at the path's exact timestamp, but TF cannot provide the transform at the requested time**, it throws the Extrapolation Error. 

To solve this I changed the local planner's global_frame to map in configuration file. This resolved the issue although it may not be ideal for local planners typically use the more continuous  odom frame.

bufeng-12, 2026-10-21:
您好！我已收到你的邮件！谢谢！          ——曾君