cartographer-project/cartographer #1963: Minimum point / second requirement in 3D?

lolhol, 2024-07-18:
Hi!

I am trying to use Unitree lidar L1 ([https://www.unitree.com/LiDAR/](https://www.unitree.com/LiDAR/)) for mapping. However I am getting horrible results at base settings. The IMU is about 200hz while the actual point cloud is about 6-7hz. All of the lidars that other people use output ~100k + points /sec while Unitree lidar L1 only outputs about 20k points /sec. Is that a problem? and if so, how can I fix it?

TiNredmc, 2025-05-02:
Hi, 

I recently bought the Unitree L1 and still waiting for the package to arrive. In the meantime I downloaded the ros bag recorded by Unitree and convert it to ros2 bag. I was able to achieve quite a good quality 3D SLAM  with some tuning. Mostly tuning the local slam. You can check it out [here](https://github.com/TiNredmc/sim_carto_l1_ros2).  
  
I made a simple ros2 package that will launch the ros2bag and the cartographer. I've to use a relatively large submap (150 to 200) to achieve a low local drift. Also due to the large submap size, the global slam was tuned to have a latency low enough that I can play the bag recording on x2 speed (on intel core i7 and 8GB ram). In my opinion, the sub map could be smaller in order to help relaxing the global slam work. You can start from my .lua configuration and work your way to tune it to suit your PC performance. I also have a plan to test the localization mode too.

bufeng-12, 2025-05-02:
您好！我已收到你的邮件！谢谢！          ——曾君

godbrigero, 2025-05-05:
> Hi,
>
> I recently bought the Unitree L1 and still waiting for the package to arrive. In the meantime I downloaded the ros bag recorded by Unitree and convert it to ros2 bag. I was able to achieve [...]

I noticed you have use_online_correlative_scan_matching set to false... doesn't that disable the whole SLAM mechanism with point clouds altogether? If so, I would imagine the position is highly based on IMU readings. Is that a problem bc of drift?

TiNredmc, 2025-05-05:
> I noticed you have use_online_correlative_scan_matching set to false... doesn't that disable the whole SLAM mechanism with point clouds altogether? If so, I would imagine the position is highly based on IMU readings. Is that a problem bc of drift?

Hi,
I disabled the fast correlative scanmatcher and used the realtime scanmather instead, but the lua values are default to the cartographer trajectory_3d.lua file.

godbrigero, 2025-05-05:
Ok! Nice! I will test on my L1 lidar soon then :) thanks for the reply.

TiNredmc, 2025-05-08:
> Ok! Nice! I will test on my L1 lidar soon then :) thanks for the reply.

If you have an update on testing please let me know! I've been struggling to tune the pure localization mode with the official ros bag recording. Due to the way that official test was done with the tilted lidar. Horizontal FoV was limited to only 180 degree. This could be the problem for the Cartographer. But other more modern SLAM like the FAST-LIO and Point-LIO are more resilient to limited FoV.

godbrigero, 2025-05-09:
Yea no problem. Another thing is that I basically can't use ROS and I really dislike C++ so I made some c++ to java bindings for the cartographer. Nothing fancy obviously just like put sensor data, get output point cloud and thats pretty much it. But like it's pretty nice to be able to use java with cartographer. 

This is kind of a side project for me and I really haven't touched cartographer since pretty much the last comment (6 months ago). So yea.

I'm probably also going to modernize some parts of the code (migrate to rust probably) since I'm not the best c++ programmer out there (I suck lol). So thats gona take even more time. 

But yea I mean I'll let you know what I have whenever I get to it :)

TiNredmc, 2025-05-10:
Oh, Java. Reminds me of the FRC. I never participated in FIRST since I'm based in Asia. But I've joined ABU Robocon for the past 2 competitions and assisting junior student at the robot club. 
It was a tradition now that most team in ABU would roll their own custom stuffs especially Electronics and Control system. Most of the time on the control side would based on ROS due to the standardized communication message. In my case the control program was written with Python (low priority such as Web GUI server) and C++ (High priority such as Lidar driver, Position controller).  I've seen another team under my professor used MATLAB to do their control.